srmainwaring/asv_wave_sim #5: Estimating values for Hydrodynamical constants

JuliusHendrix, 2020-01-24:
Hey Rhys,

I've taken a closer look at this package, and taken a look at the article by Jacques Kerner, which describes the method for simulating buoyancy you've applied and the approximations, simplifications and limitations it comes with. He also describes the quite arbitrary constants he uses for the damping of the system. These are reflected in your code as cDampL1, cDampL2, cDampR1, cDampR2 for linear and angular drag, and cPDrag1, cPDrag2, fPDrag, cSDrag1, cSDrag2, fSDrag and vRDrag for 'pressure' drag (probably the most dubious of all).

Since most of these variables have no direct link to physical processes, what did you to to find the standard values for the variables? And how would I go about determining what values would be right for my simulation? Did you eyeball the values until the behaviour seemed appropriate, or is there more thought behind them?

Kind regards,

Julius

JuliusHendrix, 2020-02-14:
Eyeballing it is

srmainwaring, 2020-03-22:
Hi Julius,

My apologies, I missed your initial comment when you first posted it. You're right, the example parameters are tuned by trial and error. I have made some initial investigations into tuning the viscosity drag parameters to results for standard hydrodynamic shapes but do not have anything to share at this point. Ideally the linear and angular parameters would be calibrated to the body's overall friction drag and the 'pressure drag' parameters to the overall form drag. The simulation effectively integrates over the contribution from each surface triangle which can be related to the integrals for global friction and form drag, but at the moment my implementation is missing a suitable length parameter along the free stream direction. I'd welcome any suggestions you may have.

Rhys

closed 2021-03-12 by JuliusHendrix