srmainwaring/asv_wave_sim #8: Speeding up wave simulations

miloknowles, 2020-09-14:
First of all, thanks for making these plugins -- they've been really helpful with my project so far!

I'm trying to simulate many floating objects at once, as in the image below:
[image]

#1: Do you have any tips for speeding up the simulation? When I publish a wave message the simulation speed drops to about 0.01 real-time. Are there any approximations/speedups that I can turn on in the plugins?

#2: This error shows up many times when I simulate a wave field on the large array of cylinders:
```
[Err] [Wavefield.cc:906] point:  -8.72634e+24 -5.07838e+24 1.38239e+23
[Err] [Wavefield.cc:908] Water patch is too small
```
Could you explain what this error means?

Thank you!

srmainwaring, 2020-09-14:
Each object has a small grid around it (water patch) which is roughly the objects axis aligned bounding box vertically projected onto the water surface. The second problem you are seeing occurs when the patch is not large enough, so one of the objects vertices projected onto the x-y plane falls outside the water patch - I thought I'd applied a fix to this from another branch but may have missed it.

The water patch provides a performance improvement for a single model with many vertices (many thousands), as it reduces the cost of searching which triangle on the water surface the vertical projection from each vertex intersects with. Unfortunately this doesn't help when you add many independent models as they each have their own water patch (even though the total vertex count may be less than for one complex model). There aren't any obvious approximations that can be tuned in the plugins at present - the depth calculation code / water patch calculation would need to be altered.

Does your array of objects need to be fully independent, or could they be coupled into a single body with joints that gave some but not full freedom? (i.e. if its going to be a swarm of robots then the answer would be no, if its supports for a single structure then maybe yes). A single body with many parts should be faster.

miloknowles, 2020-09-14:
Eventually I'm planning to connect all of those buoys together with simulated "cables" (probably just long straight pieces with ball joints at the ends to begin with). So I guess the motion of the objects will be somewhat coupled but not rigid. It sounds like it might help to make a single object composed of all the buoys, and apply the `HydroDynamics` plugin to only the top-level object?

srmainwaring, 2020-09-15:
I've reviewed how I implemented the plugin for joints, and composing the object is not going to help unfortunately. Each link in the model has its own patch, as the AABB of the composed model changes as the joints move and this was a way of ensuring that the patches supported an articulated object. In the short term you can reduce the number of objects, and I'll have to look into how to further optimise the intersection code.

miloknowles, 2020-09-16:
I appreciate the help! I'll see what I can do about simplifying the model in the meantime.

closed 2020-09-23 by miloknowles