srmainwaring/asv_wave_sim #12: Noetic Compliance

vinz-uts, 2021-08-03:
Hi, I'm trying to build the package on ROS Noetic (installed on Ubuntu 20.04), but a lot of errors are raised in the `catkin_make` process. Below an example of some errors' print:
```
/home/user/ros_ws/src/asv_wave_sim/asv_wave_sim_gazebo_plugins/src/Utilities.cc:154:44: error: ‘_sdf’ was not declared in this scope; did you mean ‘sdf’?
  154 | bool Utilities::SdfParamBool(sdf::Element& _sdf,
      |                                            ^~~~
      |                                            sdf
/home/user/ros_ws/src/asv_wave_sim/asv_wave_sim_gazebo_plugins/src/Utilities.cc:155:3: error: expected primary-expression before ‘const’
  155 |   const std::string& _paramName, const bool _defaultVal)
      |   ^~~~~
/home/user/ros_ws/src/asv_wave_sim/asv_wave_sim_gazebo_plugins/src/Utilities.cc:155:34: error: expected primary-expression before ‘const’
  155 |   const std::string& _paramName, const bool _defaultVal)
      |                                  ^~~~~
/home/user/ros_ws/src/asv_wave_sim/asv_wave_sim_gazebo_plugins/src/Utilities.cc:155:56: error: expression list treated as compound expression in initializer [-fpermissive]
  155 |   const std::string& _paramName, const bool _defaultVal)
      |                                                        ^
/home/user/ros_ws/src/asv_wave_sim/asv_wave_sim_gazebo_plugins/src/Utilities.cc:160:8: error: ‘size_t asv::Utilities::SdfParamSizeT’ is not a static data member of ‘class asv::Utilities’
  160 | size_t Utilities::SdfParamSizeT(sdf::Element& _sdf,
      |        ^~~~~~~~~
/home/user/ros_ws/src/asv_wave_sim/asv_wave_sim_gazebo_plugins/src/Utilities.cc:166:48: error: ‘_sdf’ was not declared in this scope; did you mean ‘sdf’?
  166 | double Utilities::SdfParamDouble(sdf::Element& _sdf,
      |                                                ^~~~
      |                                                sdf

```
I think the problem is related to the building process of the plugins files, maybe some change in the gcc compiler.
Had anyone tryed to run the `asv_wave_sim` package on Noetic? Thanks for the support.

nicolaje, 2021-08-31:
Hi @vinz-uts ,

I don't have any issue building the plugin on the development branch on Ubuntu20.04 with ROS Noetic.

Which branch are you building ?

vinz-uts, 2021-08-31:
Hi, thanks for your reply. I got this error when I've tried to compile the `master` branch. Next I've downloaded the other branch and with the `development` branch I haven't also issues when I build the package. But in the last week I've always the same problem: the vehicle sinking down because there is some problem with the `libHydrodynamicsPlugin`. I got a lot of this print 
```
[Err] [Wavefield.cc:896] point:  -7.63013 40.9786 0.359706
[Err] [Wavefield.cc:898] Water patch is too small
[Err] [Wavefield.cc:896] point:  -41.8836 -6.4903 0.359706
[Err] [Wavefield.cc:898] Water patch is too small
``` 
and, at the end:
```
[Err] [Model.cc:1155] Exception occured in the Init function of plugin with name[hydrodynamics] and filename[libHydrodynamicsPlugin.so]. This plugin will not run
```
I think I've do somthing wrong, but I don't understand what. I've downloaded the `asv_wave_sim-develop` package, I copy it in my ROS workspace and I've compiled it. In my package I've built my rov and I include in the `<gazebo>` tag the following code:
```
<!-- Hydrodynamics plugin -->
        <plugin name="hydrodynamics" filename="libHydrodynamicsPlugin.so">
            <!-- Wave Model -->
            <wave_model>ocean_waves</wave_model> <!-- before: ocean -->
            <!-- Hydrodynamics -->
            <damping_on>true</damping_on>
            <viscous_drag_on>true</viscous_drag_on>
            <pressure_drag_on>true</pressure_drag_on>
            <!-- Markers -->
            <markers>
                <update_rate>30</update_rate>
                <water_patch>false</water_patch>
                <waterline>false</waterline>
                <underwater_surface>false</underwater_surface>
            </markers>
        </plugin>
```
Have you any idea to help me? Thanks in advice.

srmainwaring, 2021-08-31:
@vinz-uts apologies for not getting back to you sooner. I can replicate the compile errors you are seeing on the `master` branch with ROS Noetic. These are caused by API changes between Gazebo9 and Gazebo11 and its dependencies on SDFormat.

The `feature/fft_wave` branch does compile and will run on ROS Noetic (and also ROS Melodic with custom installation of Gaazebo11). This branch does not have the wave parameters exposed in the API - so is perhaps less useful for testing a variety of conditions. 

The errors you are seeing are because the name of the ocean model in the `feature/fft_wave` branch has been shortened to `ocean`. If you change the element to read `<wave_model>ocean</wave_model>` in your model then you should see the buoyancy forces operating.

I have some maintenance to provide working branches for gazebo9, gazebo11, ignition-gazebo etc. I'll start with getting a version of master working with gazebo11.

srmainwaring, 2021-09-01:
The SDF looks correct, it may be that you have another model called `ocean` in the Gazebo resource path that is being found first. This could come from the `vrx` or `uuv_simulator` packages if they are installed. If you expand the entity tree in the Gazebo side panel is the ocean model the one you are expecting (i.e. does it have the wave plugins attached)?

vinz-uts, 2021-09-01:
You were right, there were two model with the same name. I've tried to delete the ocean model folder from `uuv_simulator` instead of renaming the ocean folder in `asv_wave_sim_gazebo` folder. Now I'm sure that the ocean models loaded is correct (I've verified the info in the side panel of Gazebo) but I get the following errors:
```
[Err] [WavefieldSampler.cc:195] point:  9.06644 40.1202 0.359706
[Err] [WavefieldSampler.cc:197] Water patch is too small
[Err] [WavefieldSampler.cc:195] point:  -7.25091 41.0352 0.359706
[Err] [WavefieldSampler.cc:197] Water patch is too small
[Err] [WavefieldSampler.cc:195] point:  -7.63013 40.9786 0.359706
[Err] [WavefieldSampler.cc:197] Water patch is too small
[Err] [WavefieldSampler.cc:195] point:  -41.8836 -6.4903 0.359706
[Err] [WavefieldSampler.cc:197] Water patch is too small
[Msg] Water patch size: 0
[Err] [Model.cc:1155] Exception occured in the Init function of plugin with name[hydrodynamics] and filename[libHydrodynamicsPlugin.so]. This plugin will not run

```
and the vehicle is still sinking. Thanks a lot for your help!

srmainwaring, 2021-09-01:
Ok - this is happening because the patch used to project the vehicle onto the water is not large enough to cover all the vertices in the vehicle mesh. (i.e. the AABB of the model projected onto the xy-plane must be covered by the patch). You can view the patches by setting `<water_patch>true</water_patch>` and pausing the simulation on start (let it evolve a couple of steps then stop before the model sinks).

Is the `feature/fft_wave` branch working with the models included in the repo? If these are working it may be that there is something amiss with the collision mesh on your model.

vinz-uts, 2021-09-01:
I've enabled the water patch view setting `<water_patch>true</water_patch>`. I see the following
[image]
is it a correct view?

srmainwaring, 2021-09-01:
I'd expect to see something more like this:

[image]

Or in wireframe mode:

[image]

(this on the `feature/fft_waves` branch). 

Are the models in `asv_wave_sim_gazebo/models` working correctly? I've just set up an Ubuntu 20.04 VM with ROS Noetic as detailed here: https://github.com/srmainwaring/asv_wave_sim/issues/13#issuecomment-909197628, and am using that to generate the samples.

vinz-uts, 2021-09-01:
The grid it looks like it is splitted under the vehicle.
[image]
I've tried to substituting the mesh files of my vehicle (both visual and geometry) with the duck model in the `asv_wave_sim_gazebo` package but also still sinking (is this the correct way to use your models?).
I don't understand why the same mesh file had not problem with Melodic version.

The `[Err] [Model.cc:1155] Exception occured in the Init function of plugin with name[hydrodynamics] and filename[libHydrodynamicsPlugin.so]. This plugin will not run` error remain and the object sinks.

srmainwaring, 2021-09-01:
If your Gazebo resource paths are set up correctly then you can start the ocean world with

```bash
$ roslaunch asv_wave_sim_gazebo ocean_world.launch verbose:=true
```

Then navigate to the *Insert* tab on the sidebar, select the duck model from the appropriate collection, and drop it into the ocean.

vinz-uts, 2021-09-07:
Hi, I've found that the main problem was in my vehicle model. My mesh model was created in mm and Gazebo considered it in meters even if I used the `scale` option in my urdf `<geometry>` tag. I did not get the error abaut the waterpatch dimension anymore but the it was still visualized in the wrong way and, in addition,  I started to have the same problem (low performance) reported by @nicolaje (I tried to use `catkin_make --cmake-args -DCMAKE_BUILD_TYPE=RelWithDebInfo` but performance doesn't improve).
In the end, I replaced my vehicle geometry with a simpler shape (just a cylinder) and now it works. My vehicle geometry is probably too complex and involves too many calculations. I'll try to make it simpler. 

Next I've switched to `develop` branch because I want to modify waves entity and others ocean's parameters.
I was doing some tests with the coefficients of friction and damping.
"""

            <cDampL1>1.0E-6</cDampL1>
            <cDampL2>1.0E-1</cDampL2>
            <cDampR1>1.0E-6</cDampR1>
            <cDampR2>1.0E-6</cDampR2>

"""
I don't understand how this parameters affect the behavior of the vehicle (the linear and quadratic coefficients seem to have the same effects). If possible, can you share a formula to better understand this relationship?

srmainwaring, 2022-05-03:
> If possible, can you share a formula to better understand this relationship?

Hi @vinz-uts, the two references to Jacques Kerner's blog at the bottom of the README contain the motivation for these parameters. They are the linear and quadratic coefficients for linear / rotational velocity dependent damping.

I'll close the issue as the main concern about building with ROS Noetic seems to be addressed.

closed 2022-05-03 by srmainwaring