srmainwaring/asv_wave_sim #13: Performance issue

nicolaje, 2021-08-31:
Hi @srmainwaring ,

first thank you a lot for this plugin, it helped significantly.

I am running your "develop" branch with Gazebo 11, and with a simple Wam-V simulation the real-time factor of Gazebo is 0.14 with only one CPU running 100%.

I have a quite decent computer (NVIDIA Quadro T2000, Corei9 with 8 cores, 32GB of RAM).

To reproduce I just clone your repository, checkout development branch, build then:

```bash
roslaunch asv_wave_sim_gazebo ocean_world.launch
```

Then from Gazebo insert the Wam-V model.

Is there something I am missing ?

Thank you !

[image]

[image]

srmainwaring, 2021-08-31:
Hi @nicolaje, I'm glad the plugin has been useful. Are you using a Wam-V model with modified collision meshes or the standard version available online?

The unmodified version will have a meshes with a high vertex count, and possible some degenerate triangles. Both will cause the simulation to slow significantly. The high vertex count results in a larger number of surfaces to sum over for the hydrodynamics forces, degenerate triangles (three points col-linear or nearly so) result in overflow conditions that are slow to compute (propagating NaNs) that are then ultimately rejected when the contributions from each triangle are aggregated.

I created a set of modified collision meshes for the Wam-V at some point and recall it running with better performance than you are seeing. It may not be on the main branch of this repo but I can search for it if that helps.

nicolaje, 2021-08-31:
Hi, thank you for the prompt answer !

I am indeed using the model of this repository.

Could you maybe point me in a direction if you need help troubleshooting?

Best regads

PS: Also, be aware that on the fft_waves & gazebo11 branches, all builds fine but when you add by hand a dock / Wam-V etc... they dont' float. I am just using it successfully with the develop branch.

srmainwaring, 2021-08-31:
> PS: Also, be aware that on the fft_waves & gazebo11 branches, all builds fine but when you add by hand a dock / Wam-V etc... they dont' float. I am just using it successfully with the develop branch.

I suspect that may be because I've changed the name of the ocean model in those branches - so the boat model is not subject to any buoyancy forces. That should be an easy fix if that's the case. If you run gazebo with the `--verbose` flag there should be messages describing what is going wrong.

I'm in the process of developing support for Ignition Gazebo at the moment which has broken a lot of my old ROS / Gazebo11 environment because of updated dependencies. I'll see if I can run up another Ubuntu VM with Noetic to investigate what is going on.

nicolaje, 2021-09-01:
Hi @srmainwaring ,

```bash
... --cmake-args -DCMAKE_BUILD_TYPE=RelWithDebInfo
```

that was it. It is the default Debug build which is slow as hell.

So **develop** is working great on my desktop when built in release node.

On the other hand, I can't get feature/fft_waves to work. I run the simulation, then if I drop a Duck or a Wam-V it sinks and I have this error message :

[image]

Is there a Paypal or something like that to thank you for your plugin ? I would love to buy you a pack of beers.

Cheers

srmainwaring, 2021-09-01:
Great! Glad it's running well now.

The issue with the model not floating has to do with the name given to the ocean model. The message is essentially saying that the hydrodynamics plugin associated with the wam-v model cannot find a wave model with the name `ocean`. I could probably improve that message!

Check the name of the ocean model in the Gazebo sidebar - I suspect it is `ocean_waves` rather than `ocean` (i.e. the sim is picking up a duplicated version of the ocean model somewhere with the wrong name). You may have to search for old / installed versions from the master build and remove them.

ricardohalfeld, 2022-04-10:
Hello. First let me take a second to congratulate on the result of your efforts. Truly beautiful plugin. This makes Gazebo a better-rounded simulator when it comes to selling a project.

Second, let me just get a little trick out there: On Gazebo's GUI, go to Physics and reduce the Real Time Update Rate. Also adjust Max Step Size to (Real Time Update Rate)^(-1). This will improve your real time simulation **at the increased risk of making it less representative**. Use with caution and make sure you understand the implications.

Cheers, folks!

srmainwaring, 2022-05-03:
I'm going to close this issue as the main problem was resolved by using the release build settings. There is of course ongoing work to improve performance generally which will be discussed in separate issues.

closed 2022-05-03 by srmainwaring