srmainwaring/asv_wave_sim #14: libHydrodynamicsPlugin.so init function Issue

ElQemmah, 2021-09-13:
Dear ASV Wave Developers! 
I am currently working with your ASV Wave Simulator plugins on ROS Noetic, and in particular exploiting the _asv_wave_sim_ devel branch, combined with uuv_simulator (in particular the libuuv_thruster_ros_plugin). I have built the urdf model of my vehicle, contaning the links, joints ecc.) combined with the urdf model of the actuators and gps/imu models. 
When I start my simulation I get the following error: 
"""
[Msg] Vertex:     24
[Msg] links:  6
[Msg] meshes: 6
[Msg] Water patch size: 2.52761
[Msg] Water patch size: 0
[Err] [Model.cc:1155] Exception occured in the Init function of plugin with name[hydrodynamics] and filename[libHydrodynamicsPlugin.so]. This plugin will not run
""" 
I have done some studies regarding your toolbox. I have become aware that the plug in at the beginning tries to create a water patch size not only for the vehicle but also for the thrusters. The reason for this is that there is a continuous joint connects the thrusters links to the vehicle link in the thruster urdf (if I change it to fixed then there is no water patch size for the thrusters, but the thruster plugin does not work). 
The problem can actually be solved by specifying the collision tag inside the urdf of the thrusters (doing so I get a real time factor of 0.01 and simulations are impossible). But since I have decided to not put any collision inside the urdf of the thrusters, I would like to solve this problem and get rid of the previous error. 

I would like to add that we successfully made all work (only on one machine) with no collision, the water patch size gets set to 0 for all thruster and no error pops up. We are not able to make it work with other machines.

srmainwaring, 2022-02-07:
Hello @ElQemmah - I have not been active in this repo for some time so apologies for the delay.

The plugin attempts to assign buoyancy to all links in the model. The easiest way to get your model working is to assign a simple collision mesh (a small cube) to each of the thrusters. This will have a minimal impact on performance as the mesh is simple, and won't require any coding on my part. If you use the visual mesh for the thrusters it will run very slowly as the vertex count will be very high for these.

srmainwaring, 2022-05-03:
Closing to lack of activity - fairly sure the issue was that the model was missing a collision mesh for thrusters.

closed 2022-05-03 by srmainwaring