srmainwaring/asv_wave_sim #66: Colcon build of Gazebo Garden version error

ConnorDTaylor, 2022-09-14:
Hi Rhys,

I have started up a VM in Ubuntu 20.04 to try and get Gazebo Garden working with ROS2 and this simulation. I got Gazebo Garden installed properly, however when I add asv_sim to the workspace which Garden in it, the package will not build and gives me the following errors.
/home/connort/gazebo_garden/src/asv_wave_sim/gz-waves/src/SubMeshWithTangents.cc:60:1: error: function ‘gz::common::SubMeshWithTangents::SubMeshWithTangents(gz::common::SubMeshWithTangents&&)’ defaulted on its redeclaration with an exception-specification that differs from the implicit exception-specification ‘’
   60 | SubMeshWithTangents::SubMeshWithTangents(SubMeshWithTangents &&_submesh)
      | ^~~~~~~~~~~~~~~~~~~
/home/connort/gazebo_garden/src/asv_wave_sim/gz-waves/src/SubMeshWithTangents.cc:79:22: error: function ‘gz::common::SubMeshWithTangents& gz::common::SubMeshWithTangents::operator=(gz::common::SubMeshWithTangents&&)’ defaulted on its redeclaration with an exception-specification that differs from the implicit exception-specification ‘’
   79 | SubMeshWithTangents &SubMeshWithTangents::operator=(SubMeshWithTangents &&_submesh)
      |                      ^~~~~~~~~~~~~~~~~~~
make[2]: *** [src/CMakeFiles/gz-waves1.dir/build.make:167: src/CMakeFiles/gz-waves1.dir/SubMeshWithTangents.cc.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:428: src/CMakeFiles/gz-waves1.dir/all] Error 2
make: *** [Makefile:163: all] Error 2
---
Failed   <<< gz-waves1 [51.8s, exited with code 2]

What would be the reason for this error?

Cheers,

Connor

ConnorDTaylor, 2022-09-14:
I think this problem may have something to do with GCC being used to compile the code instead of Clang++. It seems the noexpect = default is the culprit, I'm trying to figure out if there is a work around to this problem which isn't changing my compiler.

srmainwaring, 2022-09-14:
I did my test build on 22.04 and it seems there are a couple of compiler / library differences.

Here are the fixes for Ubuntu 20.04 with Gazebo Garden built from source: 

1. SubMeshWithTangents

Comment the move copy constructor and assignment operators (the ones with noexcept). Related to this SO question:
https://stackoverflow.com/questions/47032077/noexcept-default-compilation-error

2. TriangulatedGrid around L79

```diff
- typedef CGAL::No_constrain_intersection_tag                    Itag;
+ typedef CGAL::No_intersection_tag                                     Itag;
```

ConnorDTaylor, 2022-09-14:
Hi Rhys,

I now have the package working, but im running into a new issue which may also be caused by using Ubuntu 20.04 or it is cause by my using a VM. I should note I had to change the the default version of ogre3D (ogre2) to ogre (AKA ogre1.1) because of the VM I am using. There errors I am getting are:

[Err] [SystemLoader.cc:87] Failed to load system plugin [gz-waves1-waves-model-system] : couldn't find shared library.
[Err] [SystemLoader.cc:87] Failed to load system plugin [gz-waves1-waves-visual-system] : couldn't find shared library.

Which indicate to me that the plugins aren't being found, but I'm not sure how or where to point them. I have a feeling that these errors might be related to the switch in ogre version, in which case I will need to try to figure out a way to install ogre2 in my VM. However, I figured I would run this error by you in the mean time in case it is caused by something else. Gazebo Garden is a big shift from Gazebo9 so I don't know what I don't know.

srmainwaring, 2022-09-14:
Yes, the errors are because you are using ogre1 and the plugins require ogre2.

You can install the dependencies for Gazebo using the instructions here: https://gazebosim.org/docs/garden/install_ubuntu_src

This is the command given in the help for installing the dependencies on Ubuntu:

```bash
sudo apt -y install \
  $(sort -u $(find . -iname 'packages-'`lsb_release -cs`'.apt' -o -iname 'packages.apt' | grep -v '/\.git/') | sed '/gz\|sdf/d' | tr '\n' ' ')
```

You'll then need to rebuild `gz-rendering`, `gz-gui7`, `gz-sensors7` and `gz-sim7`.

Does the rendering work running ogre2 on your VM? The reason I ask is that on a macOS host running VM fusion you cannot use hardware acceleration on the VM because the drivers do not support OpenGL 4.2+. You can use llvmpipe, but it's practically usable for the waves simulation (good for checking it runs, but that's about it). The latter option only works on Ubuntu 22.04 because it needs a later version of mesa utils.

ConnorDTaylor, 2022-09-14:
I’ll try Ilvmpipe tomorrow when I get back into my lab, I was reading about it but I wasn’t sure if it would work. If I had my computer environment booted in 20.04 would the simulator work with ogre2? Or are you saying ogre2 is only available in Ubuntu 22.04?
 I am stuck using 20.04 due to my reliance on PX4 and i’m thinking about using this wave simulator with ROS2 for my SITL simulations. 
 
I appreciate your help Rhys, thanks a ton.

srmainwaring, 2022-09-14:
You can install ogre2 on Ubuntu 20.04, it's more a question of whether anything using the ogre2 render engine will display on an Ubuntu 20.04 VM. In my case even the basic shapes.sdf world does not render on an Ubuntu 20.04 VM (with either a hardware accelerated or llvmpipe driver).

The situation for an Ubuntu 22.04 VM is similar except that you will get something to render if you run the simulator using

```bash
LIBGL_ALWAYS_SOFTWARE=1 gz sim -v4 waves.sdf
```

but it will be very slow - unusably slow.

Your best option is to run the simulation natively on either an Ubuntu machine or (macOS if you have the patience to set up the dev environment - it runs on both Intel and Arm64 macs now). You could still spin up most of the ROS2 nodes on the VM if you have the network shared. The `ros_gz_bridge` node will need to run on the host machine though as it needs access to the gz-transport layer.

ConnorDTaylor, 2022-09-20:
Hi Rhys,

I have moved my environment to 22.04 and upon installing gazebo from binaries on the website I am still getting the following errors:

[Err] [SystemLoader.cc:87] Failed to load system plugin [gz-waves1-waves-model-system] : couldn't find shared library.
[Err] [SystemLoader.cc:87] Failed to load system plugin [gz-waves1-waves-visual-system] : couldn't find shared library.
[Err] [SystemLoader.cc:87] Failed to load system plugin [gz-waves1-hydrodynamics-system] : couldn't find shared library.
[Err] [SystemLoader.cc:87] Failed to load system plugin [gz-waves1-hydrodynamics-system] : couldn't find shared library.
[Err] [SystemLoader.cc:87] Failed to load system plugin [gz-waves1-hydrodynamics-system] : couldn't find shared library.
[Err] [SystemLoader.cc:87] Failed to load system plugin [libArduPilotPlugin.so] : couldn't find shared library.
[Err] [SystemLoader.cc:87] Failed to load system plugin [gz-waves1-hydrodynamics-system] : couldn't find shared library.
[Err] [SystemLoader.cc:87] Failed to load system plugin [gz-waves1-hydrodynamics-system] : couldn't find shared library.
[Err] [SystemLoader.cc:87] Failed to load system plugin [gz-waves1-hydrodynamics-system] : couldn't find shared library.
[Err] [SystemLoader.cc:87] Failed to load system plugin [gz-waves1-hydrodynamics-system] : couldn't find shared library.
[Err] [SystemLoader.cc:87] Failed to load system plugin [gz-waves1-hydrodynamics-system] : couldn't find shared library.
[Err] [SystemLoader.cc:87] Failed to load system plugin [gz-waves1-hydrodynamics-system] : couldn't find shared library.
[Err] [SystemLoader.cc:87] Failed to load system plugin [gz-waves1-hydrodynamics-system] : couldn't find shared library.

I figure the binary install grabs the required dependencies, so it it possible that the files arent being sourced properly?

ConnorDTaylor, 2022-09-20:
I had to add the path as:
export GZ_SIM_SYSTEM_PLUGIN_PATH=\
$HOME/colcon_ws/install/gz-waves1/lib

Because the form you sent simply gave the same error as before. However, I am now getting this permission error:

Error while loading the library [/home/connor/colcon_ws/install/gz-waves1/lib/libgz-waves1-waves-model-system.so]: libgz-waves1.so.1: cannot open shared object file: No such file or directory
[Err] [SystemLoader.cc:104] Failed to load system plugin [gz-waves1-waves-model-system] : couldn't load library on path [/home/connor/colcon_ws/install/gz-waves1/lib/libgz-waves1-waves-model-system.so].

srmainwaring, 2022-09-20:
I see, it looks like you have used the default build rather than `--merge-install`. It should not make too much difference if the environment variables are set up correctly. It still looks like a path issue though. Can you see all the plugin libraries in the folder `$HOME/colcon_ws/install/gz-waves1/lib`? I'm assuming you have run `source ./install/setup.bash`. Could you post the results of 

```
printenv | grep GZ
```

which may provide some more info on what's going on with your paths.

ConnorDTaylor, 2022-09-20:
Hi Rhys,

Im getting used to this new Gazebo and forgot to source setup.bash, I am used to only doing that when running ROS files. Sorry about that! It seems the simulation is working now.

I am seeing that the waves and items in the water keep flashing in and out of the simulator and the waves are acting sort of buggy. Have you encountered this? I am not sure if there is a way I could send a video of the environment to show clearly what I mean.

srmainwaring, 2022-09-20:
Great, it's running! I don't see rendering artefacts on my side, but this could be video card settings or something similar? You can record short mp4 files and post them in the issue (10MB or less).

ConnorDTaylor, 2022-09-20:
https://user-images.githubusercontent.com/80693505/191350880-6b654187-5474-45f5-b2ee-d0390a711a9e.mp4

Hopefully, you can see this. I am wondering if it is a graphics card issue.

srmainwaring, 2022-09-20:
You're not running two instances of the server by any chance? On macOS you need to run the client and server separately as the OS does not let ruby fork the process. On linux you can run both from the one terminal (my instructions say to run them separately because it will work on both platforms. However if you forget the -g or -s flags then you can wind up with two servers for one gui and that may cause the effect you are seeing.

ConnorDTaylor, 2022-09-20:
I am running the server and client in two different terminals, as per the instructions. I also noticed that the simulation is continuously pausing the playing, but I can rotate the simulation and move the camera around with little issue so im not so sure it is a computer hardware problem.

srmainwaring, 2022-09-20:
Hmm, very strange. Try killing all the `gz sim` processes then run both client and sever in one terminal:

```bash
gz sim -v4 waves.sdf
```

Any difference?

srmainwaring, 2022-09-20:
I don't think the warning is causing the problem - I'm seeing a similar message on the latest Gazebo build. Tracing the issue may take some work. Below is the bug report template for gz-sim. I don't need all explanation, but the system details will help, as will a print of the environment and any ogre logs.

## Environment
* OS Version: 
* Source or binary build?
  
  
* If this is a GUI or sensor rendering bug, describe your GPU and rendering system. Otherwise delete this section. 
    - Rendering plugin: [ogre | ogre2]. 
      - [ ] Sensor rendering error. 
      - [ ] GUI rendering error. 
    - EGL headless mode:
      - [ ] Running in EGL headless mode 
    - Generally, mention all circumstances that might affect rendering capabilities: 
      - [ ] running on a dual GPU machine (integrated GPU + discrete GPU)
      - [ ] running on a multi-GPU machine (it has multiple discrete GPUs)
      - [ ] running on real hardware
      - [ ] running in virtual machine
      - [ ] running in Docker/Singularity
      - [ ] running remotely (e.g. via SSH)
      - [ ] running in a cloud
      - [ ] using VirtualGL, XVFB, Xdummy, XVNC or other indirect rendering utilities
      - [ ] GPU is concurrently used for other tasks
        - [ ] desktop acceleration
        - [ ] video decoding (i.e. a playing Youtube video)
        - [ ] video encoding
        - [ ] CUDA/ROCm computations (Tensorflow, Torch, Caffe running)
        - [ ] multiple simulators running at the same time
      - [ ] other...
    - Rendering system info:
      - On Linux, provide the outputs of the following commands:
          ```bash
          LANG=C lspci -nn | grep VGA  # might require installing pciutils
          echo "$DISPLAY"
          LANG=C glxinfo -B | grep -i '\(direct rendering\|opengl\|profile\)'  # might require installing mesa-utils package
          ps aux | grep Xorg
          sudo env LANG=C X -version  # if you don't have root access, try to tell the version of Xorg e.g. via package manager
          ```
      - On Windows, run `dxdiag` and report the GPU-related information.
      - On Mac OS, open a terminal and type `system_profiler SPDisplaysDataType`. Copy the output here.
        
    - [ ] Please, attach the ogre.log or ogre2.log file from  `~/.ignition/rendering` 
    
<details>
  
```
# paste log here
```

</details>

## Description
* Expected behavior: 
* Actual behavior: 

## Steps to reproduce

1.
2.
3.

## Output

ConnorDTaylor, 2022-09-20:
OS Version: Ubuntu 22.04
Source or binary build: Binary build of Gazebo Garden
If this is a GUI or sensor rendering bug, describe your GPU and rendering system. Otherwise delete this section.
Rendering plugin: [ogre | ogre2]. Ogre2 is being used, flickering is happening to the models in the simulation.
- [  ] Sensor rendering error.
- [x] GUI rendering error.
Generally, mention all circumstances that might affect rendering capabilities:
- Running this on a PC with a single nvidia graphics card. The GPU is not being used for anything else while simulating.

Rendering System:
LANG=C lspci -nn | grep VGA
09:00.0 VGA compatible controller [0300]: NVIDIA Corporation GP107 [GeForce GTX 1050 Ti] [10de:1c82] (rev a1)

echo "$DISPLAY"
:1

LANG=C glxinfo -B | grep -i '\(direct rendering\|opengl\|profile\)'  # might require installing mesa-utils package
direct rendering: Yes
OpenGL vendor string: NVIDIA Corporation
OpenGL renderer string: NVIDIA GeForce GTX 1050 Ti/PCIe/SSE2
OpenGL core profile version string: 4.6.0 NVIDIA 470.141.03
OpenGL core profile shading language version string: 4.60 NVIDIA
OpenGL core profile context flags: (none)
OpenGL core profile profile mask: core profile
OpenGL version string: 4.6.0 NVIDIA 470.141.03
OpenGL shading language version string: 4.60 NVIDIA
OpenGL context flags: (none)
OpenGL profile mask: (none)
OpenGL ES profile version string: OpenGL ES 3.2 NVIDIA 470.141.03
OpenGL ES profile shading language version string: OpenGL ES GLSL ES 3.20

ps aux | grep Xorg
root       25130  1.7  0.4 25374120 75356 tty2   Sl+  14:16   3:14 /usr/lib/xorg/Xorg vt2 -displayfd 3 -auth /run/user/1000/gdm/Xauthority -nolisten tcp -background none -noreset -keeptty -novtswitch -verbose 3
connor    864034  0.0  0.0   9208  2384 pts/2    S+   17:25   0:00 grep --color=auto Xorg

sudo env LANG=C X -version

X.Org X Server 1.21.1.3
X Protocol Version 11, Revision 0
Current Operating System: Linux connor-Lab 5.15.0-1015-gke #18-Ubuntu SMP Sun Aug 14 05:50:07 UTC 2022 x86_64
Kernel command line: BOOT_IMAGE=/boot/vmlinuz-5.15.0-1015-gke root=UUID=cecbe26e-4ae1-457b-89a1-101e568381a5 ro quiet splash vt.handoff=7
xorg-server 2:21.1.3-2ubuntu2.1 (For technical support please see http://www.ubuntu.com/
[ogre2.log](https://github.com/srmainwaring/asv_wave_sim/files/9611053/ogre2.log)
support) 
Current version of pixman: 0.40.0

Ogre2.log is attached.

printenv output:

printenv
SHELL=/bin/bash
ROS_VERSION=2
SESSION_MANAGER=local/connor-Lab:@/tmp/.ICE-unix/25240,unix/connor-Lab:/tmp/.ICE-unix/25240
GZ_SIM_RESOURCE_PATH=/home/connor/colcon_ws/src/asv_wave_sim/gz-waves-models/models:/home/connor/colcon_ws/src/asv_wave_sim/gz-waves-models/world_models:/home/connor/colcon_ws/src/asv_wave_sim/gz-waves-models/worlds
QT_ACCESSIBILITY=1
GZ_GUI_PLUGIN_PATH=/home/connor/colcon_ws/src/asv_wave_sim/gz-waves/src/gui/plugins/waves_control/build
GZ_SIM_SYSTEM_PLUGIN_PATH=/home/connor/colcon_ws/install/gz-waves1/lib
COLORTERM=truecolor
XDG_CONFIG_DIRS=/etc/xdg/xdg-ubuntu:/etc/xdg
SSH_AGENT_LAUNCHER=gnome-keyring
XDG_MENU_PREFIX=gnome-
GNOME_DESKTOP_SESSION_ID=this-is-deprecated
PKG_CONFIG_PATH=/home/connor/colcon_ws/install/gz-waves1/lib/pkgconfig
ROS_PYTHON_VERSION=3
LANGUAGE=en_CA:en
GNOME_SHELL_SESSION_MODE=ubuntu
SSH_AUTH_SOCK=/run/user/1000/keyring/ssh
XMODIFIERS=@im=ibus
DESKTOP_SESSION=ubuntu
[... 19 lines cut ...]
GNOME_TERMINAL_SCREEN=/org/gnome/Terminal/screen/2f75a9e6_baf2_445b_9f42_eab35c94a05e
COLCON_PREFIX_PATH=/home/connor/colcon_ws/install
LESSCLOSE=/usr/bin/lesspipe %s %s
XDG_SESSION_CLASS=user
PYTHONPATH=/opt/ros/humble/lib/python3.10/site-packages:/opt/ros/humble/local/lib/python3.10/dist-packages
TERM=xterm-256color
LESSOPEN=| /usr/bin/lesspipe %s
USER=connor
GNOME_TERMINAL_SERVICE=:1.478
DISPLAY=:1
SHLVL=1
QT_IM_MODULE=ibus
LD_LIBRARY_PATH=/home/connor/colcon_ws/install/gz-waves1/lib:/opt/ros/humble/opt/rviz_ogre_vendor/lib:/opt/ros/humble/lib/x86_64-linux-gnu:/opt/ros/humble/lib:/usr/local/cuda/lib64
XDG_RUNTIME_DIR=/run/user/1000
ROS_LOCALHOST_ONLY=0
XDG_DATA_DIRS=/usr/share/ubuntu:/usr/share/gnome:/usr/local/share/:/usr/share/:/var/lib/snapd/desktop
PATH=/opt/ros/humble/bin:/usr/local/cuda/bin:/opt/ros/melodic/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin$:/home/connor/.local/bin
GDMSESSION=ubuntu
DBUS_SESSION_BUS_ADDRESS=unix:path=/run/user/1000/bus
GZ_VERSION=garden
ROS_DISTRO=humble
OLDPWD=/home/connor
_=/usr/bin/printenv

As previously mentioned, the issue is the models in the simulation flicker in and out. The simulation also repeatedly pauses and starts.

srmainwaring, 2022-09-21:
@ConnorDTaylor I can't see anything obvious that might cause the issue you're seeing. Your graphics card has 4GB RAM which should be fine. I can't see how many cores and what RAM you PC has: it may be CPU or RAM bound? 

Have you tried running any of the Gazebo examples (there are a number in `gz-sim/examples/worlds`). Some of them are graphics / CPU intensive (for example the ones testing sensors). At the moment it's difficult to tell if it's a problem with my plugins in particular or something else in your set up.

One other thing to try is exporting `GZ_IP=127.0.0.1`, it could be a transport issue with the client lagging because pose updates from the server are taking too long - although I'm just speculating.

ConnorDTaylor, 2022-09-21:
So I tried he GZ_IP, which didnt seem to make much difference however when trying to simualtion today I am getting these warnings which I think is in reference to the objects in the water:
[GUI] [Wrn] [Component.hh:144] Trying to serialize component with data type [St6vectorIdSaIdEE], which doesn't have `operator<<`. Component will not be serialized.
[GUI] [Wrn] [Model.hh:49] Unable to serialize sdf::Model
[GUI] [Wrn] [Model.hh:49] Unable to serialize sdf::Model

Then when I get rid of the objects in the water I am left with these warnings:
[GUI] [Wrn] [Application.cc:845] [QT] file::/Gazebo/GazeboDrawer.qml:147:3: QML Dialog: Binding loop detected for property "implicitHeight"
[GUI] [Wrn] [Application.cc:845] [QT] file::/EntityTree/EntityTree.qml:148:7: QML ToolButton: Detected anchors on an item that is managed by a layout. This is undefined behavior; use Layout.alignment instead.
[GUI] [Wrn] [Application.cc:845] [QT] file::/WorldStats/WorldStats.qml:53:3: QML RowLayout: Binding loop detected for property "x"
[GUI] [Wrn] [Application.cc:845] [QT] file::/EntityContextMenuPlugin/EntityContextMenuPlugin.qml:52:3: QML RenderWindowOverlay: Detected anchors on an item that is managed by a layout. This is undefined behavior; use Layout.alignment instead.
[GUI] [Wrn] [Application.cc:845] [QT] file::/EntityContextMenuPlugin/EntityContextMenuPlugin.qml:67:3: QML EntityContextMenu: Detected anchors on an item that is managed by a layout. This is undefined behavior; use Layout.alignment instead.
[GUI] [Wrn] [EntityComponentManager.cc:1845] Component type [4916660844221477451] has not been registered in this process, so it can't be deserialized.
[GUI] [Wrn] [Model.hh:98] Unable to deserialize sdf::Model
Warning [Utils.cc:129] [/sdf/plugin[@name="gz::sim::systems::WavesVisual"]:<data-string>:L1]: XML Element[plugin], child of element[sdf], not defined in SDF. Copying[plugin] as children of [sdf].

ConnorDTaylor, 2022-09-21:
UPDATE: I am still getting the warnings detailed above, however after noticing the wave sim kept reverting back to the waves.sdf I thought the computer may have been having some sort of memory problem. So I restarted and now the wave sim seems to be working smoothly! I will let you know if any other issues arise but I think it is working as of now.

srmainwaring, 2022-09-21:
> So I restarted and now the wave sim seems to be working smoothly!

Can't believe I missed the universal tech support fix! Glad to hear you're up and running at last.

bigbigyan, 2025-02-21:
I have the same error. How do you fix it, by building gz-garden from source ?
/home/zy/Program/DINO-V2/Wave_sim/src/asv_wave_sim/gz-waves/src/SubMeshWithTangents.cc:63:1: error: function ‘gz::common::SubMeshWithTangents::SubMeshWithTangents(gz::common::SubMeshWithTangents&&)’ defaulted on its redeclaration with an exception-specification that differs from the implicit exception-specification ‘’
   63 | SubMeshWithTangents::SubMeshWithTangents(SubMeshWithTangents&& _submesh)
      | ^~~~~~~~~~~~~~~~~~~
/home/zy/Program/DINO-V2/Wave_sim/src/asv_wave_sim/gz-waves/src/SubMeshWithTangents.cc:83:22: error: function ‘gz::common::SubMeshWithTangents& gz::common::SubMeshWithTangents::operator=(gz::common::SubMeshWithTangents&&)’ defaulted on its redeclaration with an exception-specification that differs from the implicit exception-specification ‘’
   83 | SubMeshWithTangents& SubMeshWithTangents::operator=(
      |                      ^~~~~~~~~~~~~~~~~~~
make[2]: *** [src/CMakeFiles/gz-waves1.dir/build.make:167: src/CMakeFiles/gz-waves1.dir/SubMeshWithTangents.cc.o] Error 1
make[2]: *** Waiting for unfinished jobs....

srmainwaring, 2025-02-21:
@bigbigyan if you are using Ubuntu 20.04, then yes I think the answer given in https://github.com/srmainwaring/asv_wave_sim/issues/66#issuecomment-1247141546 applies.

bigbigyan, 2025-02-21:
Thank you very much, I will try it.