srmainwaring/asv_wave_sim #68: Noetic Setup

MoffKalast, 2022-09-24:
Quick question, is there a way to get this built with Noetic or should I stop before waste a few hours?

Looking at the various branches like it's either Melodic with 9 or Humble with Garden. What's so incompatible about Fortress/11?

srmainwaring, 2022-09-25:
The master branch does not depend on ROS. The previous version for Gazebo11 did, and worked with Melodic. That branch may work with Noetic as there was only a weak dependency, mainly on the catkin build system.
The master branch targets Garden as that version has full support for macOS / Metal. There are large namespace and library changes between Fortress and Garden that would require significant effort to backport, so it’s just Garden I’m afraid.

MoffKalast, 2022-09-25:
Ah I see, there is in fact a feature/gazebo11 branch that I missed. I'll check that out. As for Garden vs Fortress, yeah that's what I figure might be the issue there.

Thanks.

srmainwaring, 2022-09-26:
@MoffKalast I'll close this. If there's any maintenance work required to get the gazebo11 branch running add an issue and I'll take a look at it.

closed 2022-09-26 by srmainwaring