srmainwaring/asv_wave_sim #111: Waves visable to me but not to sensors?

LFS-Engineering, 2022-12-12:
Hi, 
I managed to get the waves working and although I was struggling to get a boat model working I just set the collisions to a a box the same size as the lower boat and now its able to move with the waves. I am trying to test sensors such as LiDAR's in the simulation and they work fine and I can see the models that I place on the waves moving. But the waves themself are not showing as visible objects? is there a way to turn them into objects that sensors can see ?

srmainwaring, 2022-12-12:
Hi @LFS-Engineering, using range and lidar sensors to sense waves for the legacy versions is discussed in: https://github.com/srmainwaring/asv_wave_sim/issues/64.

In summary: the range and lidar sensors in Gazebo9 / Gazebo11 will not see the waves. In Gazebo Garden they will (there is no code in the plugin that specifically enables this - it's a by product of  the Gazebo Garden render engine and the new sensor plugins).

The rendering pipeline changed considerably after Gazebo11, so there is no way to back port this functionality unfortunately. 

Figure 1. Lidar returns in Gazebo Garden.

[image]

arjo129, 2023-01-22:
You can disable waves from showing in LiDAR. See: https://github.com/osrf/mbzirc/pull/111

srmainwaring, 2023-01-23:
Thanks @arjo129 - good to know. Closing this as I don't think there is further action to take at this stage.

closed 2023-01-23 by srmainwaring

Mathosmls, 2025-10-02:
Hi, I have the same issue and the solution provided by @arjo129 doesn't seem to work : 

The waves : 

```
<?xml version='1.0' ?>
<sdf version="1.6">
  <model name="waves">
    <static>true</static>
<plugin
        filename="gz-waves1-waves-model-system"
        name="gz::sim::systems::WavesModel">
      <static>0</static>
      <update_rate>30</update_rate>
      <wave>
        <!-- `fft` wave parameters -->
        <algorithm>fft</algorithm>
        <tile_size>256</tile_size>
        <cell_count>256</cell_count>
        <wind_speed>1.5</wind_speed>
        <wind_angle_deg>135</wind_angle_deg>
        <steepness>2.</steepness>
      </wave>
    </plugin>

    <link name="base_link">
      <visual name="visual">
        <laser_retro>0.0000000000001</laser_retro>
        <geometry>
          <mesh>
            <uri>materials/mesh_L256m_N256.dae</uri>
          </mesh>
        </geometry>
        <!-- 
          Uncomment material if using DYNAMIC_GEOMETRY which uses PBS
          rather than a custom shader
        -->
        <material>
          <ambient>0.0 0.05 0.8 0.92</ambient>
          <diffuse>0.0 0.1 0.7 0.92</diffuse>
          <specular>0.7 0.7 0.7 0.92</specular>
          <pbr>
            <metal>
              <albedo_map>materials/water.jpg</albedo_map>
              <normal_map>materials/wave_normals.dds</normal_map>
              <environment_map>materials/skybox_lowres.dds</environment_map>
              <roughness>0.3</roughness>
              <metalness>0.1</metalness>
            </metal>
          </pbr>
        </material>

        <!-- \todo remove duplicate params; populate from service instead -->
        <plugin
            filename="gz-waves1-waves-visual-system"
            name="gz::sim::systems::WavesVisual">
         ....

        </plugin>
        <visibility_flags>8</visibility_flags>
      </visual>
    </link>
  </model>
</sdf>

```

My lidar : 

```
<?xml version="1.0" ?>
<sdf version="1.7">
  <model name="lidar_os1-64">

    <link name="base_link">
      <inertial>
        <mass>0.1</mass>
      </inertial>

      <visual name="visual">
        <geometry>
          <mesh>
            <uri>models://lidar_os1-64/meshes/visual.dae</uri>
          </mesh>
        </geometry>
      </visual>

      <sensor name="lidar_sensor" type="gpu_ray">
        <pose>0 0 0 0 0 0</pose>
        <update_rate>10</update_rate>
        <visualize>true</visualize>
        <lidar>
          <scan>
            <horizontal>
              <samples>1024</samples>
              <resolution>1</resolution>
              <min_angle>-3.14159</min_angle>
              <max_angle>3.14159</max_angle>
            </horizontal>
            <vertical>
              <samples>64</samples>
              <resolution>1</resolution>
              <min_angle>-0.393</min_angle> <!-- -45°/2 en rad -->
              <max_angle>0.393</max_angle>  <!-- 45°/2 en rad -->
            </vertical>
          </scan>
          <range>
            <min>0.05</min>
            <max>120</max>
            <resolution>0.001</resolution>
          </range>
          <noise>
            <type>gaussian</type>
            <mean>0.0</mean>
            <stddev>0.0</stddev>
          </noise>
          <visibility_mask>55</visibility_mask>
        </lidar>
      </sensor>
    </link>

  </model>
</sdf>
```

The result :

[image]

The visibility_flags tag seems to have no effects.
I'm on gazebo garden and I don't have a gpu