srmainwaring/asv_wave_sim #114: Perscribing wave period for regular waves

ConnorDTaylor, 2022-12-19:
Hi Rhys,

I have been using this simulator to good effect and a part of my testing requires using sensor data to determine wave characteristics. I have developed a ROS2 node to determine the wave period, and it seems to be working properly. I have noticed a discrepancy though. 

What I noticed is this: If I set the period of regular, sinusoidal waves to 3 then the true wave period is closer to 4 seconds. Similarly, if I set it to 4 the period seems to be closer to 6 seconds. Is this an error which how the period is determined? If I use a timer and track the period myself I find it to be 6 seconds as well, so I'm confused why the true period is a second or two off of what I am giving the model as an argument for the period. 

Cheers,

Connor

srmainwaring, 2022-12-19:
Hi Conner, that's interesting. My first check would be to verify your test node is counting Gazebo simulation time rather than clock time. If you can make the node available I'll take a look.

The regular sinusoid wave model is fairly straight forward and you can see the conversion from input period to angular frequency in the code which is then used to calculate the wave elevation:

eta(x, y, t) = amplitude * cos( k * (x * sin(beta) + y * sin(beta) - w * t)

T = wave period
w = angular temporal frequency = 2 pi / T
k = angular spatial wave number = sqrt(w) /  gravity
beta = wave incidence angle
t = sim_time (!= wall clock time)

ConnorDTaylor, 2022-12-20:
I don't have a specific node for this, to check this I decided to make a buoy and just track its position then I plotted that z-position over time. I also have a sonar sensor which is tracking height and I used to buoy to compare the behavior. From this, I am finding that the period, which I declared to be 3 seconds, is actually around 4.5s-5s. Here is the plot I made to show this. 

If you place a buoy in the water and track its position as I have I'm interested to know if this is also the case for you. I'm not sure what would cause this behavior.

[image]

srmainwaring, 2022-12-21:
@ConnorDTaylor in your plots how are you calculating the time? i.e. when you measure the wave elevation are you using the Gazebo Sim time to measure the time or do you use something like the system clock time (via std::chrono).

You must use the Gazebo Sim time to get correct results - I'm not sure if there is a direct interface for this. You could subscribe to the `/clock` message on `gz-transport`, or write a Gazebo system plugin and query the `UpdateInfo` structure.

A quick check whether you or not you are using the simulation time would be to alter the RTF by adding the element:

```xml
<physics name="1ms" type="ignore">
  <max_step_size>0.001</max_step_size>
  <real_time_factor>1.0</real_time_factor>
</physics>
```

and setting `<real_time_factor>` to different values.

srmainwaring, 2023-01-22:
@ConnorDTaylor I've investigated this a little further, and am fairly sure that your issue has to do with the clock you are using to plot your graphs.

This commit  https://github.com/srmainwaring/asv_wave_sim/commit/0af5e7debf3fa53c8f5ff778ad944002e1e4f01a contains a couple of test models of a WEC device that I'm using for some other analysis - the 1/10 scale version can be used as a wave height sensor (low mass so little lag and other inertial effects - a small object bobs like a cork with the water surface). I'm using the Pose system plugin to output the pose and some Python tools I've written to capture the Gazebo messages and output to ROS and a sample to file. As far as I can tell the wave period observed is the same as specified in the wave model (period 6s, amplitude 2m).

Figure: 1/10 scale version of the RM3 WEC device for measuring wave height.  
 
[image]

Figure: time series plot of the float's z position versus the simulation time. 

[image]

closed 2023-01-22 by srmainwaring

ConnorDTaylor, 2023-01-22:
Hi Rhys,

Sorry about going dark on this, I forgot I raised this issue. You are right, I didn't realize that my simulation was running at 70%-80% speed due to hardware limitations. This is why all of my calculated wave periods are around 30% longer than I thought they would be. So you are correct, it is a clock issue. Sorry for not mentioning this earlier and thank you for looking into it more I really appreciate it.

Cheers,
Connor