srmainwaring/asv_wave_sim #132: not able to build due to cgal error

smvaraprasad, 2023-02-01:
I have followed the instructions in your github [readme.md](https://github.com/srmainwaring/asv_wave_sim/blob/master/README.md)

input command:
[image]

output:
[image]
[image]
  
  
  expected output:
  successful build
 
  
  looked at these links:
   https://github.com/CGAL/cgal/issues/6647
   https://github.com/microsoft/vcpkg/issues/23098
   did not understand what to do.
   
   followup:
   on trying to build again, I didnt get error this time, but build is not finished yet. it is at 40% after 40 mins. [image]
   
   how much time, it should take normally?
    
  configuration:
  gazebo garden
  wsl2 ubuntu 22.04 jammy
  i5 9th gen, 8gb ram, windows 10 (updated version)

srmainwaring, 2023-02-01:
Hi @smvaraprasad, I have not attempted to build on WSL, and I am not familiar enough with the environment to know whether there would be differences compared to running natively on an Ubuntu machine.

The build takes under a minute on my mac, and a bit over a minute on an Ubuntu Kinetic (22.10) VM (assigned 4 processors and 8GB RAM).

The versions of CGAL I see on these systems are:

macOS:

```bash
% brew info cgal
==> cgal: stable 5.5.1 (bottled)
Computational Geometry Algorithms Library
https://www.cgal.org/
/usr/local/Cellar/cgal/5.5.1 (3,681 files, 38.4MB) *
  Poured from bottle on 2023-01-08 at 23:00:17
From: https://github.com/Homebrew/homebrew-core/blob/HEAD/Formula/cgal.rb
License: GPL-3.0-or-later
```

Ubuntu:

```
$ apt search libcgal-dev 
Sorting... Done
Full Text Search... Done
libcgal-dev/kinetic,now 5.5-1 amd64 [installed]
  C++ library for computational geometry (development files)
```

Either way, you should be able to ignore the warning or follow this advice (https://stackoverflow.com/questions/71713439/what-is-cgaldata-file-path) and add

```cmake
set(CGAL_DATA_DIR ".")
```

to CMakeLists.txt as the `CGAL_DATA_DIR` is not used by the code (no calls to the function that requires it).

srmainwaring, 2023-02-02:
@smvaraprasad - I can confirm that this project builds and runs on WSL2.

[image]

There are a couple of updates needed for the instructions:

There is a build dependency on TBB (should not be needed now, but seems to be left in CMakeLists.txt):

```bash
sudo apt install libtbb-dev
```

When running you also need to specify the system plugin location:

```bash
export GZ_SIM_SYSTEM_PLUGIN_PATH=\
$HOME/gz_ws/install/lib
```

System details:

- Win11 in a VMware Fusion 13 VM. 4 cores and 16GB RAM, 8GB shared graphics memory.
- WLS2 with Ubuntu 22.04.

```bash
$ uname -a
Linux Win11 5.15.79.1-microsoft-standard-WSL2 #1 SMP Wed Nov 23 01:01:46 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux
```

Gazebo Garden was installed following the instructions here: https://gazebosim.org/docs/garden/install_ubuntu.
Before the install I ran a `sudo apt update && sudo apt upgrade` to get everything current.

You also need the dev tools from the install from source page: https://gazebosim.org/docs/garden/install_ubuntu_src.

The build took about 3-4 mins on WLS2.

I'm not sure what might be causing your problems. If you try building in the source directory and post a full build log it may be clearer (switch off the test build to speed things up and reduce dependencies):

```bash
cd $HOME/gz_ws/src/asv_wave_sim/gz-waves
mkdir build && cd build
cmake .. -DCMAKE_BUILD_TYPE=RelWithDebInfo -DCMAKE_CXX_STANDARD=17 -DBUILD_TESTING=OFF
```

srmainwaring, 2023-02-03:
@smvaraprasad closing as verified to work. I suspect this is a case of updating your WSL2 / Ubuntu install rather than a problem with this project. Feel free to post build logs if you are unable to resolve.

closed 2023-02-03 by srmainwaring

smvaraprasad, 2023-02-08:
[error_asv_srmainwaring.txt](https://github.com/srmainwaring/asv_wave_sim/files/10681409/error_asv_srmainwaring.txt)
This is the total error, when tried to run command " colcon build --symlink-install --merge-install --cmake-args \
-DCMAKE_BUILD_TYPE=RelWithDebInfo \
-DBUILD_TESTING=ON \
-DCMAKE_CXX_STANDARD=17 "

[error_asv_srmainwaring_2.txt](https://github.com/srmainwaring/asv_wave_sim/files/10681448/error_asv_srmainwaring_2.txt)
This is the total error, when tried to run command "cmake .. -DCMAKE_BUILD_TYPE=RelWithDebInfo -DCMAKE_CXX_STANDARD=17 -DBUILD_TESTING=OFF
make -j4" 

system:
ubuntu 22.04
=>cgal version is 5.4

I have followed all of your instructions, excuse the delay. As I switched from wsl to ubuntu (dual boot).

srmainwaring, 2023-02-08:
@smvaraprasad thanks for the build logs. Both builds completed without errors (a number of warnings, but these don't prevent the build completing). The line `stderr output` refers to the stream of warnings emitted by the compiler / linker during the build, it does not mean the build failed. 

```bash
# from error_asv_srmainwaring.txt

...

Finished <<< gz-waves1 [4min 21s]
                           
Summary: 1 package finished [4min 21s]
  1 package had stderr output: gz-waves1
```

You should be able to run the simulation if you build using colcon in the workspace directory.

I'll look into eliminating the warnings that apply to the wave plugins. The other warnings for various gz libs will need to be corrected upstream.

smvaraprasad, 2023-02-08:
[image]
There is a distorted sky, static water with no waves and as i started the simulation, boat, duck etc sunk. apparently there are some plugins missing according to the  log, these plugins are mentioned in the end of readme.md under the title of changes. Was i supposed to make the changes?
[error_asv_srmainwaring_3.txt](https://github.com/srmainwaring/asv_wave_sim/files/10684279/error_asv_srmainwaring_3.txt)

smvaraprasad, 2023-02-13:
The issue is still the same, output still looks the same. logs are still the same. system is ubuntu 22.04 jammy. it is saying it is unable to find the plugins through some library search paths, kindly let me know how to correct those paths if there is any error. I have followed all your instructions mentioned.
- and the mentioned graphics driver issue, how to correct that?

smvaraprasad@smvaraprasad-IdeaPad-L340-15IRH-Gaming:~/gz_ws/src$ export | grep GZ_SIM_RESOURCE_PATH
declare -x GZ_SIM_RESOURCE_PATH="/home/smvaraprasad/gz_ws/src/asv_wave_sim/gz-waves-models/models:/home/smvaraprasad/gz_ws/src/asv_wave_sim/gz-waves-models/world_models:/home/smvaraprasad/gz_ws/src/asv_wave_sim/gz-waves-models/worlds"

[error_asv_srmainwaring_4.txt](https://github.com/srmainwaring/asv_wave_sim/files/10718727/error_asv_srmainwaring_4.txt)

srmainwaring, 2023-02-13:
It looks like your library path is still not correct. Please run:

```bash
printenv | grep GZ_SIM
```

and paste the result into your reply.

srmainwaring, 2023-02-13:
> no output... tried from both gz_ws/src and home directory

nothing from 

```bash
env | grep GZ_SIM
```

either? Then you have not set up the environment correctly. The two variables `GZ_SIM_SYSTEM_PLUGIN_PATH` and `GZ_SIM_RESOURCE_PATH` need to include the locations of all plugins and resources respectively according to the instructions in the README.

smvaraprasad, 2023-02-13:
looks like i have to run the export command everytime, now i ran it again. after this i tried to simulate , still the output and log files are same.

[printenv_output_asv_srmainwaring.txt](https://github.com/srmainwaring/asv_wave_sim/files/10720379/printenv_output_asv_srmainwaring.txt)

srmainwaring, 2023-02-13:
> looks like i have to run the export command everytime

Yes, it needs to be run in every terminal you use to run the simulation. This is why you might want to add the export to the ~/.bashrc or set up shell script containing the exports you can source when running simulations.

The environment is incomplete:

```bash
smvaraprasad@smvaraprasad-IdeaPad-L340-15IRH-Gaming:~/gz_ws/src$ printenv | grep GZ_VERSION
GZ_VERSION=garden
smvaraprasad@smvaraprasad-IdeaPad-L340-15IRH-Gaming:~/gz_ws/src$ printenv | grep GZ_SIM
GZ_SIM_RESOURCE_PATH=/home/smvaraprasad/gz_ws/src/asv_wave_sim/gz-waves-models/models:/home/smvaraprasad/gz_ws/src/asv_wave_sim/gz-waves-models/world_models:/home/smvaraprasad/gz_ws/src/asv_wave_sim/gz-waves-models/worlds
```

The last command should include an entry for `GZ_SIM_SYSTEM_PLUGIN_PATH` as well, but doesn't indicating you have not set this variable correctly.

smvaraprasad, 2023-02-13:
Was this supposed to be done manually? Sorry but I didn't find this in the readme.md . Can you help me set this env variable?

srmainwaring, 2023-02-13:
> Was this supposed to be done manually? Sorry but I didn't find this in the readme.md . Can you help me set this env variable?

https://github.com/srmainwaring/asv_wave_sim#set-environment-variables

srmainwaring, 2023-02-13:
Good to know it's working.

The WAM-V is set up to be controller by ArduPilot, so it's looking for the ArduPilot plugin. The warning can be safely ignored...

I'm not sure what is going on with the sky rendering. As mentioned earlier, if the error persists when the waves model and plugin is removed then its an issue with the core Gazebo render engine rather than this plugin. There may be a conflict with the waves visual, but I cannot replicate it.

The other errors (the collision errors) have to do with the DART physics engine in `gz-physics6`. They are not particular to this plugin. They can be safely ignored - I believe the Gazebo team are aware of them and there may be an issue already to deal with that.

smvaraprasad, 2023-02-13:
Okay, Thank you for your time!