srmainwaring/asv_wave_sim #137: Stay on waves when you reach the end of the map

therishidesai, 2023-04-28:
When I run the asv_wave_sim and the boat reaches the end of the water surface it just keeps on floating in air. Is there a way to have the entire map be water and when it reaches the end it just loops back?

srmainwaring, 2023-04-28:
Hi @therishidesai, the physics has period boundary conditions. i.e.  the physics system sees the waves repeat with dimensions determined by the tile size. The rendering system places the number of tiles requested in the plugin XML and will display this only. For the PBR renderer it is quite expensive to tile using the current render engine. For the custom shader it is less so, and you can tile very large regions for little extra cost, but the visuals are lower fidelity.

If by 'loops back' you mean period boundary conditions, then yes this is what is implemented.

therishidesai, 2023-04-28:
So when we drive the boat and it is on the water everything is working as expected, but the water only extends to a certain portion of the map and so when we go off the water the waves stop acting on it. How do we guarantee that we are always on the water? When we try making the water bigger it starts slowing things down a lot, so that's why I mentioned looping back to the middle of the water when reach the end.

Here are pictures showing what I'm talking about:
[image]
[image]

The first picture shows the boat on the water, and the second pictures shows how we reach the edge of the water and are just floating in air

srmainwaring, 2023-04-28:
@therishidesai no matter where the boat is on the map it will be subject to waves. This is because in Gazebo the physics (which run on the server) and the visuals (what you see in the GUI) are distinct, and the wave physics is periodic over the entire plane.

What this means is that if your wave tile has dimensions 256 m x 256 m then the wave at (x, y) and the wave at (x + n * 256, y + m * 256) where n and m are integers will be the same. So the boat is not floating in the air, it is subject to the hydrodynamic forces you'd expect for periodic boundary conditions. If you monitor the boat pose it will be apparent is still subject to wave motion.

The visuals are a different matter. For the PBR render engine it turns out to be very expensive to tile the map by repeating the same visual displaced across the entire map. The reasons are technical, and if you would like the full details I can explain, but the quick answer is that with the current way it works you need to copy all the mesh vertices for every tile to from the CPU to the GPU every time you update a frame. As the number of tiles increases the performance degrades quadratically (if tiling in both directions).

If you need a very large map you can decrease the mesh resolution (not great for small boats), keep your vehicle to a region of interest, or use the non-PBR render engine which will tile the visuals more efficiently because it uses a different technique. The `ocean_waves` model (https://github.com/srmainwaring/asv_wave_sim/blob/master/gz-waves-models/world_models/ocean_waves/model.sdf) uses this approach and the waves can be extended across a large region by changing the `<tiles_x>` and `<tiles_y>` elements.

therishidesai, 2023-04-29:
Ah ok, I guess we didn’t notice the boat was having forced acting on it but it was probably due to the waves being small. Thanks for the information on this.

closed 2023-04-29 by srmainwaring