srmainwaring/asv_wave_sim #138: Questions about the effect of buoyancy on the water surface

gaya7502, 2023-05-12:
Hello, I use your plug-in, but when I set the coordinate z of the ocean model to -25, the objects I placed (like boxes) will still float on the original horizontal plane, where do I need to set it to adjust What about the horizontal position?

srmainwaring, 2023-05-13:
Hello @gaya7502, there is currently an assumption that the wave model will be anchored at the origin:

https://github.com/srmainwaring/asv_wave_sim/blob/b3eac206e3a73e4b41010278bba59f29452fed12/gz-waves/src/Wavefield.cc#L92-L96

This could be relaxed but would need some work to propagate the pose of the tile wrt the world frame.

gaya7502, 2023-05-15:
Sorry, I think I may not have made it clear, the plugin version I am currently using is Wave Sim gazebo 9, and the content of Wavefield is not consistent with what you describe.

srmainwaring, 2023-05-15:
I see, you're using branch [gazeb9](https://github.com/srmainwaring/asv_wave_sim/tree/gazebo9).

The cause however is similar (although the comments are not so clear about the assumptions):

https://github.com/srmainwaring/asv_wave_sim/blob/fef63188f5c8d18da00bd704cc4c07c1831563c2/asv_wave_sim_gazebo_plugins/src/Wavefield.cc#L802-L826

srmainwaring, 2023-05-18:
I've made a note to address this in #140, but that will only be for Gazebo Garden. The branches for Gazebo 9 and 11 are retained for legacy use but are no longer being updated.

closed 2023-05-18 by srmainwaring