srmainwaring/asv_wave_sim #139: Cannot successfully build package

fbenti, 2023-05-15:
Ubuntu 22
Ros2 Humble 

The colcon build command that you specified returns the following error:

[stderr.log](https://github.com/srmainwaring/asv_wave_sim/files/11475370/stderr.log)

fbenti, 2023-05-15:
Also, unrelated, is there any parameter to change the dimension of the area?

srmainwaring, 2023-05-15:
The ArduPilot plugin needs to be built separately - you can comment the plugin SDF in the WAM-V out unless you're planning to use AP for control. The instructions for the ArduPilot plugin are here: https://ardupilot.org/dev/docs/sitl-with-gazebo.html

The dimensions are controlled in the wave model SDF. The README has details: https://github.com/srmainwaring/asv_wave_sim#waves-model-and-visual-plugins

srmainwaring, 2023-05-18:
> Looks like it was necessary to restart the pc, now I can build it correctly.

The build issue appears to be resolved - closing.

closed 2023-05-18 by srmainwaring