srmainwaring/asv_wave_sim #146: mumbles_head.sdf crashing

tejalbarnwal, 2023-08-08:
I am on Ubuntu 22.04 using gazebo garden with ROS2 Humble.
When using `gz sim -v4 -r mumbles_head.sdf`, though GUI does open up, the server crashes with the following error
```
terminate called after throwing an instance of 'Ogre::ItemIdentityException'
 what():  OGRE EXCEPTION(4:ItemIdentityException): A texture with name 'HeightMapTex(6)' already exists. (Real tex name: 
'HeightMapTex(6)') in TextureGpuManager::createTexture at ./OgreMain/src/OgreTextureGpuManager.cpp (line 385)
```

srmainwaring, 2023-08-08:
Hi @tejalbarnwal, interesting. I don't see the error on macO running Gazebo Garden (there is no dependency on ROS so the choice of distro shouldn't be affect the behaviour).

This is what the world should look like:

Figure: mumbles head world
[image]

There are two methods used to render the water, the first uses the PBR shaders and updates the vertices on the CPU and pushes them to the GPU, the second uses a custom shader which uses textures as displacement maps. The textures for the vertex displacements and tangent space are updated on the CPU and then pushed to the GPU where the shader applies the shifts to a regular grid. The latter method scales better as it can be tiled without copying the vertices for each tile - however something seems to be going wrong in your case when the texture is applied repeatedly over each tile.

You should only need the element

```xml
<include>
  <pose>0 0 0 0 0 0</pose>
  <uri>model://ocean_waves</uri>
</include>
```
to replicate the issue, and can reduce the number of tiles in the `ocean_waves` model to run faster.

Does the problem disappear for a single tile? 

I'll see if I can replicate on a Ubuntu VM.

srmainwaring, 2023-08-24:
> Did you get time to look into the above issue?

I'm afraid not. If you are able to use `DYNAMIC_GEOMETRY` for your requirements that should provide a working alternative.

srmainwaring, 2023-08-25:
Hi @tejalbarnwal - I have been able to reproduce on an Ubuntu 22.04 VM (VMware) with software rendering. I think the issue is caused by this upstream issue https://github.com/gazebosim/gz-sim/issues/1647.

What happens is the visual plugin is loaded twice, the second time causes the error on the texture load. On macOS I am using this patch to prevent the second load: https://github.com/gazebosim/gz-sim/commit/e251548da84de0ac3d0c271817273c5f5aafcb99

You could apply the patch and see if that addresses the issue you are seeing.

I could apply a work-around here, or a check (and skip the second load), but really the problem is upstream.

tejalbarnwal, 2023-08-27:
>You could apply the patch and see if that addresses the issue you are seeing.

Hey, I tried the patch and then it worked! Thanks.

> If you are able to use DYNAMIC_GEOMETRY for your requirements, that should provide a working alternative.

Yeah, I experimented with DYNAMIC_GEOMETRY, it's just the RTF deteriorates as I try to increase the area covered by tile

closed 2023-08-27 by tejalbarnwal