srmainwaring/asv_wave_sim #148: Could not find shared library Hydrodynamics

iliabaranov, 2023-10-02:
Hello all,

I followed the install guide, but unfortunately three strange things happen.

The first is that the sky in gazebo looks strange, like a moire pattern.
Secondly, the above error shows up, which leads me to the third error, all objects immediately sink to the bottom on sim start.

I get a similar error on the client complaining about missing WavesVisual

This is on a freshly installed Ubuntu 22.04 system, gazebo Garden. Everything seemed to compile and exports were done. 

What are next steps to help debug?

Thanks!

srmainwaring, 2023-10-02:
Hi @iliabaranov, this seems like a potential combination of things - perhaps GPU compatibility for the sky rendering and unfound plugins.

> all objects immediately sink to the bottom on sim start.

> I get a similar error on the client complaining about missing WavesVisual

These seem like an env variable issues.

Could you run the server and gui separately with full logging and post the console output as text into a comment. If you have some details about the device that may help as well the ogre2 log (`~/.gz/rendering/ogre2.log `).

Server:

```bash
gz sim -v4 -s -r waves.sdf
```

Terminal:

```bash
gz sim -v4 -g
```

srmainwaring, 2023-10-27:
@iliabaranov did you manage to get your system running? I'll close this issue as there has not been recent activity, but feel free to reopen if you have follow up questions.

closed 2023-10-27 by srmainwaring

iliabaranov, 2023-11-03:
Hey, sorry for the delay

I followed the guide, and env variables seem populated, for example:
```
iliabara@Metis:~/gz_ws$ echo $GZ_GUI_PLUGIN_PATH 
:/home/iliabara/gz_ws/src/asv_wave_sim/gz-waves/src/gui/plugins/waves_control/build
iliabara@Metis:~/gz_ws$ echo $GZ_SIM_RESOURCE_PATH 
:/home/iliabara/gz_ws/src/asv_wave_sim/gz-waves-models/models:/home/iliabara/gz_ws/src/asv_wave_sim/gz-waves-models/world_models:/home/iliabara/gz_ws/src/asv_wave_sim/gz-waves-models/worlds

```

Ogre2.log attached: 
[ogre2.log](https://github.com/srmainwaring/asv_wave_sim/files/13246556/ogre2.log)

iliabaranov, 2023-11-03:
Ah, found it for the hydrodynamics plugin at least:

The readme says:
```
export GZ_SIM_SYSTEM_PLUGIN_PATH=\
$GZ_SIM_SYSTEM_PLUGIN_PATH:\
$HOME/gz_ws/install/lib
```
Whereas if I change this to the following, it works well
```
export GZ_SIM_SYSTEM_PLUGIN_PATH=\
$GZ_SIM_SYSTEM_PLUGIN_PATH:\
$HOME/gz_ws/src/install/lib
```

iliabaranov, 2023-11-03:
Still get the bizarre sky though...

srmainwaring, 2023-11-06:
> Whereas if I change this to the following, it works well

@iliabaranov did you run the colcon build command from the workspace/src directory? This would explain why you need to modify the path. The issue with the sky appears unrelated.