srmainwaring/asv_wave_sim #152: Performance issues using FFT waves

mvanlobensels, 2024-01-31:
Hi,

First of all thank you for your work.

I am noticing performance issues when using the FFT waves. The Gazebo starts to stutter a lot. When using the sinusoidal waves, the simulation is butter smooth. My assumption is that the FFT waves are more expensive to compute? Is there anything that can be done to increase the performance?

mvanlobensels, 2024-01-31:
Changing both `tile_size` and `cell_count` in WavesVisual fixed it

closed 2024-01-31 by mvanlobensels

srmainwaring, 2024-01-31:
Hi @mvanlobensels - yes, the simulation is demanding because I'm calculating the FFT on the CPU 2x each frame because of the way Gazebo separates the server and client logic. Furthermore in Garden there was a bug where the visuals plugins would be loaded twice costing an extra calc (this may have been fixed). Reducing the resolution will help but at the cost of wave definition: 256 -> 128 cell count may make the difference.