srmainwaring/asv_wave_sim #153: plugins arent found in the asv_wave_sim/install/lib path

kwikius, 2024-02-02:
Operating system Ubuntu22.04. 
After build plugins exist but aren't found in the asv_wave_sim/install/lib path, but are found if you change to the asv_wave-sim/build/gz-waves1/lib path.

**_There is a very similar problem in asv_sim_**, which I solved by switching on the symlinks executable flag for the symlinks in the install/lib directory.  This didn't work so well in asv_wave-sim, some plugins still failed to load, but I did solve it by just copying the files over from asv_wave-sim/build/gz-waves1/lib path to asv_wave_sim/install/lib .

I guess it has something to do with file permissions but I don't know what

srmainwaring, 2024-02-06:
Hi @kwikius, are you using the `--merge-install` flag with the colcon build? If you don't include that flag colcon will install the libraries for each package to `{workspace_dir}/install/{pkg_name}/lib`.

The screenshot below is from an Ubuntu 22.04.04 LTS VM showing the contents of the ./install/lib folder after a build using the commands given in the README.

[image]

kwikius, 2024-02-07:
I forgot to answer about the build. I am on Ubuntu, didn't use a VM and copy-pasted the build commands straight from https://github.com/srmainwaring/asv_wave_sim#ubuntu-1

srmainwaring, 2024-02-08:
Hi @kwikius, still cannot reproduce the file permission flags you are seeing. After a clean rebuild of the project workspace I see:

[image]

which has the executable flags set. The only thing I can suggest is that your user permissions may be incorrect, or you've installed some tools using sudo? (sudo installed python packages perhaps?). It's very odd. If you create a small scratch colcon cmake project in a new workspace and use similar build settings do you see the same output? You could try dropping the `--symlink-install` flag and see if that changes anything.

I am fairly certain that this is not a bug in the project CMakeLists.txt, as the cmake settings are very standard and I cannot reproduce your file permissions on either macOS or Ubuntu. It has has not been reported as an issue by other users.

kwikius, 2024-02-09:
I don't think it is the build which is the problem. 
Your install/lib output flags are the same as mine. 

The problem is that my version of gazebo-sim does not seem to find the dlls in asv_wave_sim/install/libs, though they exist but does find them in asv_wave_sim/build/gz-waves1/lib

Note however, that though your symlinks have executable flags set, the actual dll file they point to libgz-waves1-dynamic-geometry-system.so.1.0.0 does not, which may be the problem with gazebo finding the dlls in my version.

The difference between the 2 directories seems to be that in asv_wave_sim/build/gz-waves1/lib libgz-waves1-dynamic-geometry-system.so.1.0.0 has the executable flags set, but in asv_wave_sim/install/libs it does not. 

Below shows the output looking at each directory after a fresh build, where you can see the different state of the flags in the .so.1.0.0 files in each directory.

```
andy@lenovo:~/temp/gz_test/asv_wave_sim/install/lib$ ls -l
total 67000
drwxrwxr-x 10 andy andy     4096 Feb  9 08:39 cmake
lrwxrwxrwx  1 andy andy       41 Feb  9 08:39 libgz-waves1-dynamic-geometry-system.so -> libgz-waves1-dynamic-geometry-system.so.1
lrwxrwxrwx  1 andy andy       45 Feb  9 08:39 libgz-waves1-dynamic-geometry-system.so.1 -> libgz-waves1-dynamic-geometry-system.so.1.0.0
-rw-r--r--  1 andy andy  5458344 Feb  9 08:32 libgz-waves1-dynamic-geometry-system.so.1.0.0
lrwxrwxrwx  1 andy andy       38 Feb  9 08:39 libgz-waves1-hydrodynamics-system.so -> libgz-waves1-hydrodynamics-system.so.1
lrwxrwxrwx  1 andy andy       42 Feb  9 08:39 libgz-waves1-hydrodynamics-system.so.1 -> libgz-waves1-hydrodynamics-system.so.1.0.0
-rw-r--r--  1 andy andy 11663848 Feb  9 08:33 libgz-waves1-hydrodynamics-system.so.1.0.0
lrwxrwxrwx  1 andy andy       33 Feb  9 08:39 libgz-waves1-rendering-ogre2.so -> libgz-waves1-rendering-ogre2.so.1
lrwxrwxrwx  1 andy andy       37 Feb  9 08:39 libgz-waves1-rendering-ogre2.so.1 -> libgz-waves1-rendering-ogre2.so.1.0.0
-rw-r--r--  1 andy andy  8874552 Feb  9 08:39 libgz-waves1-rendering-ogre2.so.1.0.0
lrwxrwxrwx  1 andy andy       27 Feb  9 08:39 libgz-waves1-rendering.so -> libgz-waves1-rendering.so.1
lrwxrwxrwx  1 andy andy       31 Feb  9 08:39 libgz-waves1-rendering.so.1 -> libgz-waves1-rendering.so.1.0.0
-rw-r--r--  1 andy andy  2941168 Feb  9 08:34 libgz-waves1-rendering.so.1.0.0
lrwxrwxrwx  1 andy andy       17 Feb  9 08:39 libgz-waves1.so -> libgz-waves1.so.1
lrwxrwxrwx  1 andy andy       21 Feb  9 08:39 libgz-waves1.so.1 -> libgz-waves1.so.1.0.0
-rw-r--r--  1 andy andy 27585344 Feb  9 08:31 libgz-waves1.so.1.0.0
lrwxrwxrwx  1 andy andy       36 Feb  9 08:39 libgz-waves1-waves-model-system.so -> libgz-waves1-waves-model-system.so.1
lrwxrwxrwx  1 andy andy       40 Feb  9 08:39 libgz-waves1-waves-model-system.so.1 -> libgz-waves1-waves-model-system.so.1.0.0
-rw-r--r--  1 andy andy  4318328 Feb  9 08:33 libgz-waves1-waves-model-system.so.1.0.0
lrwxrwxrwx  1 andy andy       37 Feb  9 08:39 libgz-waves1-waves-visual-system.so -> libgz-waves1-waves-visual-system.so.1
lrwxrwxrwx  1 andy andy       41 Feb  9 08:39 libgz-waves1-waves-visual-system.so.1 -> libgz-waves1-waves-visual-system.so.1.0.0
-rw-r--r--  1 andy andy  7744920 Feb  9 08:38 libgz-waves1-waves-visual-system.so.1.0.0
drwxrwxr-x  2 andy andy     4096 Feb  9 08:39 pkgconfig
andy@lenovo:~/temp/gz_test/asv_wave_sim/install/lib$ cd ../
andy@lenovo:~/temp/gz_test/asv_wave_sim/install$ cd ../
andy@lenovo:~/temp/gz_test/asv_wave_sim$ cd build/gz-waves1/lib
andy@lenovo:~/temp/gz_test/asv_wave_sim/build/gz-waves1/lib$ ls -l
total 77800
-rw-rw-r-- 1 andy andy 10954744 Feb  9 08:26 libgtest.a
-rw-rw-r-- 1 andy andy   108742 Feb  9 08:31 libgtest_main.a
lrwxrwxrwx 1 andy andy       41 Feb  9 08:32 libgz-waves1-dynamic-geometry-system.so -> libgz-waves1-dynamic-geometry-system.so.1
lrwxrwxrwx 1 andy andy       45 Feb  9 08:32 libgz-waves1-dynamic-geometry-system.so.1 -> libgz-waves1-dynamic-geometry-system.so.1.0.0
-rwxrwxr-x 1 andy andy  5458344 Feb  9 08:32 libgz-waves1-dynamic-geometry-system.so.1.0.0
lrwxrwxrwx 1 andy andy       38 Feb  9 08:33 libgz-waves1-hydrodynamics-system.so -> libgz-waves1-hydrodynamics-system.so.1
lrwxrwxrwx 1 andy andy       42 Feb  9 08:33 libgz-waves1-hydrodynamics-system.so.1 -> libgz-waves1-hydrodynamics-system.so.1.0.0
-rwxrwxr-x 1 andy andy 11663848 Feb  9 08:33 libgz-waves1-hydrodynamics-system.so.1.0.0
lrwxrwxrwx 1 andy andy       33 Feb  9 08:39 libgz-waves1-rendering-ogre2.so -> libgz-waves1-rendering-ogre2.so.1
lrwxrwxrwx 1 andy andy       37 Feb  9 08:39 libgz-waves1-rendering-ogre2.so.1 -> libgz-waves1-rendering-ogre2.so.1.0.0
-rwxrwxr-x 1 andy andy  8874552 Feb  9 08:39 libgz-waves1-rendering-ogre2.so.1.0.0
lrwxrwxrwx 1 andy andy       27 Feb  9 08:34 libgz-waves1-rendering.so -> libgz-waves1-rendering.so.1
lrwxrwxrwx 1 andy andy       31 Feb  9 08:34 libgz-waves1-rendering.so.1 -> libgz-waves1-rendering.so.1.0.0
-rwxrwxr-x 1 andy andy  2941168 Feb  9 08:34 libgz-waves1-rendering.so.1.0.0
lrwxrwxrwx 1 andy andy       17 Feb  9 08:31 libgz-waves1.so -> libgz-waves1.so.1
lrwxrwxrwx 1 andy andy       21 Feb  9 08:31 libgz-waves1.so.1 -> libgz-waves1.so.1.0.0
-rwxrwxr-x 1 andy andy 27585344 Feb  9 08:31 libgz-waves1.so.1.0.0
lrwxrwxrwx 1 andy andy       36 Feb  9 08:33 libgz-waves1-waves-model-system.so -> libgz-waves1-waves-model-system.so.1
lrwxrwxrwx 1 andy andy       40 Feb  9 08:33 libgz-waves1-waves-model-system.so.1 -> libgz-waves1-waves-model-system.so.1.0.0
-rwxrwxr-x 1 andy andy  4318328 Feb  9 08:33 libgz-waves1-waves-model-system.so.1.0.0
lrwxrwxrwx 1 andy andy       37 Feb  9 08:38 libgz-waves1-waves-visual-system.so -> libgz-waves1-waves-visual-system.so.1
lrwxrwxrwx 1 andy andy       41 Feb  9 08:38 libgz-waves1-waves-visual-system.so.1 -> libgz-waves1-waves-visual-system.so.1.0.0
-rwxrwxr-x 1 andy andy  7744920 Feb  9 08:38 libgz-waves1-waves-visual-system.so.1.0.0

```

kwikius, 2024-04-15:
Closing this. I think it may have been a problem with my installation

closed 2024-04-15 by kwikius