srmainwaring/asv_wave_sim #154: Using the plugins in ignition Gazebo

NarimanNiknejad, 2024-02-07:
I am trying to use the plugins developed here in ignition gazebo but I had no luck. I know that I need to rub them on Gazebo Garden in order for them to work but is there any way I can run the plugins on ignition Gazebo as well? The reason I am trying to do that is because we have developed all the codes for a differential drive robot for ignition Gazebo and I am not sure how to port them to Gazebo Garden. Whenever I have the plugins in my world file and I launch it I get the following error and it does not load the plugin. 

[ruby $(which ign) gazebo-1] Library [~/gz_ws/install/lib/libgz-waves1-waves-visual-system.so] does not export any plugins. The symbol [IgnitionPluginHook] is missing, or it is not externally visible.

Thank you in advance!

srmainwaring, 2024-02-07:
Hi @NarimanNiknejad, no I'm afraid the plugins will not run on versions of Gazebo before Gazebo Garden (`gz-sim7`). The reason is that there are major namespace changes between Ignition Fortress and Gazebo Garden which affect includes, namespaces and macros. It was too much work to support the previous version, and in any case the only support for macOS in Ignition Fortress was available in unreleased custom branches.

The [migration guide](https://gazebosim.org/docs/garden/migration_from_ignition) may help you porting from Ignition Fortress to Gazebo Garden.

closed 2024-02-07 by srmainwaring

NarimanNiknejad, 2024-02-07:
Hi, thank you so much! Just wanted to make sure! I appreciate it!