srmainwaring/asv_wave_sim #155: Could not find shared library

AntonioViscomi, 2024-02-09:
Hi all,
I'm trying to run the simulator on gazebo harmonic (ubuntu 22.4). I've done the build with test and no test failed.
When I do <code>'gz sim -v4 -s -r waves.sdf' </code> and <code>gz sim -v4 -g</code> some errors shows up. <br>

In the server terminal I get:

> **[Err] [SystemLoader.cc:92] Failed to load system plugin [gz-waves1-waves-model-system] : Could not find shared library.**
> **[Err] [SystemLoader.cc:92] Failed to load system plugin [gz-waves1-waves-visual-system] : Could not find shared library.**
> **[Err] [SystemLoader.cc:92] Failed to load system plugin [gz-waves1-hydrodynamics-system] : Could not find shared library.**
> **[Err] [SystemLoader.cc:92] Failed to load system plugin [gz-waves1-hydrodynamics-system] : Could not find shared library.**
> **[Err] [SystemLoader.cc:92] Failed to load system plugin [gz-waves1-hydrodynamics-system] : Could not find shared library.**
> **[Err] [SystemLoader.cc:92] Failed to load system plugin [ArduPilotPlugin] : Could not find shared library.**
> **[Err] [SystemLoader.cc:92] Failed to load system plugin [gz-waves1-hydrodynamics-system] : Could not find shared library.**
> **[Err] [SystemLoader.cc:92] Failed to load system plugin [gz-waves1-hydrodynamics-system] : Could not find shared library.**
> **[Err] [SystemLoader.cc:92] Failed to load system plugin [gz-waves1-hydrodynamics-system] : Could not find shared library.**
> **[Err] [SystemLoader.cc:92] Failed to load system plugin [gz-waves1-hydrodynamics-system] : Could not find shared library.**
> **[Err] [SystemLoader.cc:92] Failed to load system plugin [gz-waves1-hydrodynamics-system] : Could not find shared library.**
> **[Err] [SystemLoader.cc:92] Failed to load system plugin [gz-waves1-hydrodynamics-system] : Could not find shared library.**
> **[Err] [SystemLoader.cc:92] Failed to load system plugin [gz-waves1-hydrodynamics-system] : Could not find shared library.**

While in the client:

> **[GUI] [Err] [SystemLoader.cc:92] Failed to load system plugin [gz-waves1-waves-visual-system] : Could not find shared library.**

I've followed the wiki procedure (also installed gzdev).

srmainwaring, 2024-02-09:
@AntonioViscomi - this looks like your `GZ_SIM_SYSTEM_PLUGIN_PATH` is not correct, although cannot tell without more information about your system and environment.

AntonioViscomi, 2024-02-09:
> @AntonioViscomi - this looks like your `GZ_SIM_SYSTEM_PLUGIN_PATH` is not correct, although cannot tell without more information about your system and environment.
It solved the problem.
I'm moving my first steps in gazebo and apparently I've built the whole thing in gz_ws/src instead of gz_ws.
My fault, thanks a lot.

AntonioViscomi, 2024-02-09:
This solved the problem.
I'm taking my first steps in gazebo and it seems I've built the whole thing in gz_ws/src instead of gz_ws.
My fault, thanks a lot.