srmainwaring/asv_wave_sim #156: What kind of wave model is used here ?

vbalaji21, 2024-02-26:
Can you please let me know what kind of wave model is used here ?
And also is the wave only visual or can it have physics based interaction with objects too ?

srmainwaring, 2024-02-26:
Hi @vbalaji21, the wave models are described in the last two documents of the acknowledgments section in the README. Both visual and physics interactions are implemented.

You can see some additional example uses and slide here:
- https://github.com/ArduPilot/SITL_Models/blob/master/Gazebo/docs/BlueBoat.md
- https://discuss.ardupilot.org/t/sailboat-support/32060/895?u=rhys
- https://community.gazebosim.org/t/community-meeting-maritime-jun-2022/1472/6
- https://docs.google.com/presentation/d/1JXwWMPPVT7y03Vr6cWtrwAwyvdysmg3NW9ZmI276dMY/edit#slide=id.p

srmainwaring, 2024-02-26:
> I am looking into it now and i am finding it very useful.

Great. The model is Tessendorf waves with some attention paid to ensure the modelled spectral properties are accurate (this is described in some detail in Curtis Mobley's article).

The buoyancy physics is described in Kerner's article series. To get a proper hydrodynamics model requires some additional treatment - discussed further here: https://github.com/srmainwaring/asv_wave_sim/issues/69. This may or may not matter depending on your requirements.

vbalaji21, 2024-02-26:
@srmainwaring another quick question. The system requirements mentioned here is Ubuntu 22.
I have a system of Ubunutu 20.04 and Gazebo 11. Is it possible to use your work in my configuration ?

srmainwaring, 2024-02-26:
> @srmainwaring another quick question. The system requirements mentioned here is Ubuntu 22. I have a system of Ubunutu 20.04 and Gazebo 11. Is it possible to use your work in my configuration ?

No, they will not work with Gazebo 11. That version is EOL and will be discontinued in 2025. These plugins only work with the new versions of Gazebo (Garden and Harmonic).

srmainwaring, 2024-02-29:
Question answered - closing.

closed 2024-02-29 by srmainwaring