srmainwaring/asv_wave_sim #157: Assistance with PX4 Boat Simulation Integration

PerFrivik, 2024-02-28:
Dear Mr. Mainwaring,

I want to express my gratitude for creating such an incredible repository. It has been an enjoyable experience using it in a standalone mode. However, I've encountered some difficulties while integrating it with my PX4 boat simulation.

In PX4 architecture, worlds and vehicles are managed separately, allowing vehicles to be spawned into an independently created world file. I attempted to add a world with the wave plugin and a separate boat equipped with the hydrodynamics plugin to the PX4 GZ Sim. Unfortunately, the boat seems to pass through the waves, possibly because hydroData.size() in hydrodynamics.cc is zero. This leads me to suspect that the plugins might require both the boat and waves to be included in the same .world file. Could you confirm if this is the case, or if I might be misunderstanding the setup?

For context, we successfully used asv_wave_sim in our Gazebo classic boat simulation, where an older version of the plugin was compatible:

<plugin name='usv_dynamics_wamv_dynamics_plugin' filename='libgazebo_usv_dynamics_plugin.so'>
      <bodyName>base_link</bodyName>
      <waterLevel>0</waterLevel>
      <waterDensity>997.8</waterDensity>
      <xDotU>0.0</xDotU>
      <yDotV>0.0</yDotV>
      <nDotR>0.0</nDotR>
      <xU>51.3</xU>
      <xUU>72.4</xUU>
      <yV>102.6</yV>
      <yVV>0.0</yVV>
      <zW>500.0</zW>
      <kP>50.0</kP>
      <mQ>50.0</mQ>
      <nR>400.0</nR>
      <nRR>0.0</nRR>
      <hullRadius>0.213</hullRadius>
      <boatWidth>2.4</boatWidth>
      <boatLength>4.9</boatLength>
      <length_n>2</length_n>
      <wave_model>ocean_waves</wave_model>
    </plugin>
    
   I would love to hear if it would be possible to run on PX4, thank you for the help! 
   
   Best regards, 
   
   Per

srmainwaring, 2024-02-28:
Hi @PerFrivik, good to hear that you’ve found the plug-ins useful. It should be possible to set a PX4 example up as you describe. I have not tried configuring for PX4 but have a number of models running with Ardupilot as the controller. I don’t think there is anything that requires the model to be included in the world file rather than spawned.

is there a model in particular you are looking to set up? If you have a repo you can share and a branch with the current set up I may be able to take a look.

PerFrivik, 2024-02-28:
Hey @srmainwaring thank you for the fast reply! I'll try again tomorrow and provide you with a branch and more details. 

Just a small recap, incase you see something obviously wrong that I am doing: 

I took the wam-v model from your repository and basically copied the URDF into PX4 (for now) and linked the plugin binaries using environment variables. Then in a .world file I inlcude the wave model. This means I have a URDF of the wam-v, which I then import to the .world file when I start PX4, and I think this importing step might cause  the issue with the initialization of the hydrodynamics plugin, as the waves appear nicely, but the boat just falls through them. 

Again, thank you so much for open sourcing such a cool project! I will make sure that when this is in a clean state to give you all the credit you deserve on the PX4 side!

PerFrivik, 2024-03-01:
Hey, I managed to fix it! The issue was that PX4 renames your model, so it is not the robot name in the SDF, but rather it gets the PX4 vehicle name from the airframe and that's why it did not work. Thank you for the help! I'll keep working with it and update you in the coming days.

closed 2024-03-01 by PerFrivik