srmainwaring/asv_wave_sim #158: [Err] [SystemLoader.cc:92] Failed to load system plugin [gz-waves1-hydrodynamics-system] : Could not find shared library.

dlaboy, 2024-02-28:
Hello! I'm having problems loading the hydrodynamics plugin, any idea what the problem could be?

srmainwaring, 2024-02-28:
@dlaboy it’s most likely you have not set up the GZ environment variables correctly, but you will need to provide more information in order to diagnose your problem.

Should include platform, OS version, Gazebo version, output logs from gz sim -v4.

dlaboy, 2024-02-29:
I'm running it on macOS Sonoma version 14.3.1. Gazebo Harmonic, these are the output logs from gz sim -v4

[image]

dlaboy, 2024-02-29:
I apologize, I think I solve this particular problem, colcon build aborted the particular gz-waves1 directory so it wasn't creating install/lib executable files. I'm having another problem know hahah. which I will post next.

dlaboy, 2024-02-29:
Ones I run gz sim-v4 -s -r catamaran_waves.sdf, this error appears

dlaboy@Diegos-MacBook-Pro ~ % gz sim -v4 -s -r catamaran_waves.sdf
[Msg] Gazebo Sim Server v8.1.0
[Msg] Loading SDF world file[/Users/dlaboy/Desktop/college/roboboat_stuff/repos/gz_ws/src/SITL_Models/Gazebo/worlds/catamaran_waves.sdf].
[Msg] Serving entity system service on [/entity/system/add]
[Dbg] [Physics.cc:860] Loaded [gz::physics::dartsim::Plugin] from library [/opt/homebrew/Cellar/gz-physics7/7.0.0_4/lib/gz-physics-7/engine-plugins/libgz-physics-dartsim-plugin.dylib]
[Dbg] [SystemManager.cc:74] Loaded system [gz::sim::systems::Physics] for entity [1]
[Msg] Create service on [/world/waves/create]
[Msg] Remove service on [/world/waves/remove]
[Msg] Pose service on [/world/waves/set_pose]
[Msg] Pose service on [/world/waves/set_pose_vector]
[Msg] Light configuration service on [/world/waves/light_config]
[Msg] Physics service on [/world/waves/set_physics]
[Msg] SphericalCoordinates service on [/world/waves/set_spherical_coordinates]
[Msg] Enable collision service on [/world/waves/enable_collision]
[Msg] Disable collision service on [/world/waves/disable_collision]
[Msg] Material service on [/world/waves/visual_config]
[Msg] Material service on [/world/waves/wheel_slip]
[Dbg] [SystemManager.cc:74] Loaded system [gz::sim::systems::UserCommands] for entity [1]
[Dbg] [Sensors.cc:527] Configuring Sensors system
[Dbg] [Sensors.cc:446] SensorsPrivate::Run
[Dbg] [Sensors.cc:422] SensorsPrivate::RenderThread started
[Dbg] [SystemManager.cc:74] Loaded system [gz::sim::systems::Sensors] for entity [1]
[Dbg] [Sensors.cc:262] Waiting for init
[Dbg] [SystemManager.cc:74] Loaded system [gz::sim::systems::SceneBroadcaster] for entity [1]
[Dbg] [SystemManager.cc:74] Loaded system [gz::sim::systems::Imu] for entity [1]
[Dbg] [SystemManager.cc:74] Loaded system [gz::sim::systems::NavSat] for entity [1]
[Err] [SystemLoader.cc:92] Failed to load system plugin [asv_sim2-anemometer-system] : Could not find shared library.
WARNING: All log messages before absl::InitializeLog() is called are written to STDERR
E0000 00:00:1709209400.230898   15253 descriptor_database.cc:633] File already exists in database: gz/msgs/time.proto
F0000 00:00:1709209400.231043   15253 descriptor.cc:2122] Check failed: GeneratedDatabase()->Add(encoded_file_descriptor, size) 
*** Check failure stack trace: ***
    @        0x10090ea7c  absl::lts_20230802::log_internal::LogMessage::SendToLog()
    @        0x10090e580  absl::lts_20230802::log_internal::LogMessage::Flush()
    @        0x10090ed8c  absl::lts_20230802::log_internal::LogMessageFatal::~LogMessageFatal()
    @        0x10090eda4  absl::lts_20230802::log_internal::LogMessageFatal::~LogMessageFatal()
    @        0x10495516c  google::protobuf::DescriptorPool::InternalAddGeneratedFile()
    @        0x1049d0880  google::protobuf::(anonymous namespace)::AddDescriptors()
    @        0x1049d0868  google::protobuf::(anonymous namespace)::AddDescriptors()
    @        0x1049d0868  google::protobuf::(anonymous namespace)::AddDescriptors()
    @        0x1049d0868  google::protobuf::(anonymous namespace)::AddDescriptors()
    @        0x1049d08b4  google::protobuf::internal::AddDescriptorsRunner::AddDescriptorsRunner()
    @        0x1893f5a24  ___ZZNK5dyld46Loader25findAndRunAllInitializersERNS_12RuntimeStateEENK3$_0clEv_block_invoke
    @        0x18943b0f4  ___ZNK5dyld313MachOAnalyzer18forEachInitializerER11DiagnosticsRKNS0_15VMAddrConverterEU13block_pointerFvjEPKv_block_invoke.202
    @        0x18942e668  ___ZNK5dyld39MachOFile14forEachSectionEU13block_pointerFvRKNS0_11SectionInfoEbRbE_block_invoke
    @        0x1893d52fc  dyld3::MachOFile::forEachLoadCommand()
    @        0x18942d6a0  dyld3::MachOFile::forEachSection()
    @        0x189430188  dyld3::MachOFile::forEachInitializerPointerSection()
    @        0x18943ade8  dyld3::MachOAnalyzer::forEachInitializer()
    @        0x1893f1b38  dyld4::Loader::findAndRunAllInitializers()
    @        0x1893f7f70  dyld4::JustInTimeLoader::runInitializers()
    @        0x1893f1f24  dyld4::Loader::runInitializersBottomUp()
    @        0x1893f1f00  dyld4::Loader::runInitializersBottomUp()
    @        0x1893f1f00  dyld4::Loader::runInitializersBottomUp()
    @        0x1893f1f00  dyld4::Loader::runInitializersBottomUp()
    @        0x1893f5ab0  dyld4::Loader::runInitializersBottomUpPlusUpwardLinks()::$_1::operator()()
    @        0x1893f20f0  dyld4::Loader::runInitializersBottomUpPlusUpwardLinks()
    @        0x1894107ac  dyld4::APIs::dlopen_from()
    @        0x10062e380  gz::plugin::Loader::Implementation::LoadLib()
    @        0x10062e144  gz::plugin::Loader::LoadLib()
    @        0x10109bb74  gz::sim::SystemLoaderPrivate::InstantiateSystemPlugin()
    @        0x10109b758  gz::sim::v8::SystemLoader::LoadPlugin()
    @        0x10109ea68  gz::sim::v8::SystemManager::LoadPlugin()
    @        0x101080210  gz::sim::v8::SimulationRunner::LoadPlugins()
    @        0x101064144  gz::common::EventT<>::Signal<>()
    @        0x10104b0b8  gz::sim::v8::EventManager::Emit<>()
    @        0x101049fa4  gz::sim::v8::SdfEntityCreator::CreateEntities()
    @        0x101014d94  gz::sim::v8::LevelManager::LoadActiveEntities()
    @        0x1010142bc  gz::sim::v8::LevelManager::UpdateLevelsState()
[... 11 lines cut ...]
    @        0x1fa707144  rb_vm_exec
    @        0x1fa5e61d0  ruby_exec_node
    @        0x1fa5e606c  ruby_run_node
    @        0x10027353c  main
[... cut ...]
#3    Object "libsystem_platform.dylib", at 0x189789a23, in _sigtramp + 55
#2    Object "libgz-tools2-backward.dylib", at 0x100353377, in backward::SignalHandling::sig_handler(int, __siginfo*, void*) + 19
#1    Object "libgz-tools2-backward.dylib", at 0x1003533db, in backward::SignalHandling::handleSignal(int, __siginfo*, void*) + 71
#0    Object "libgz-tools2-backward.dylib", at 0x10035351b, in backward::StackTraceImpl<backward::system_tag::darwin_tag>::load_from(void*, unsigned long, void*, void*) + 43
zsh: abort      gz sim -v4 -s -r catamaran_waves.sdf

dlaboy, 2024-02-29:
These are the values for GZ_*

dlaboy@Diegos-MacBook-Pro ~ % env | grep GZ
GZ_SIM_SYSTEM_PLUGIN_PATH=/Users/dlaboy/Desktop/college/roboboat_stuff/repos/gz_ws/src/ardupilot_gazebo/build:/Users/dlaboy/Desktop/college/roboboat_stuff/repos/gz_ws/install/lib:
GZ_SIM_RESOURCE_PATH=:/Users/dlaboy/Desktop/college/roboboat_stuff/repos/gz_ws/src/SITL_Models/Gazebo/models:/Users/dlaboy/Desktop/college/roboboat_stuff/repos/gz_ws/src/SITL_Models/Gazebo/worlds:/Users/dlaboy/Desktop/college/roboboat_stuff/repos/gz_ws/src/asv_wave_sim/gz-waves-models/models:/Users/dlaboy/Desktop/college/roboboat_stuff/repos/gz_ws/src/asv_wave_sim/gz-waves-models/world_models:/Users/dlaboy/Desktop/college/roboboat_stuff/repos/gz_ws/src/asv_wave_sim/gz-waves-models/worlds
GZ_VERSION=garden
GZ_GUI_PLUGIN_PATH=$GZ_GUI_PLUGIN_PATH:~/Desktop/college/roboboat_stuff/repos/gz_ws/src/asv_wave_sim/gz-waves/src/gui/plugins/waves_control/build

dlaboy, 2024-02-29:
Okk I will try this

dlaboy, 2024-02-29:
1. I'm using a binary installation
2. sensors seem to run well on example for iris_runway.sdf from ardupilot_gazebo. 

dlaboy@Diegos-MBP ~ % gz sim -v4 -s -r iris_runway.sdf
[Msg] Gazebo Sim Server v8.1.0
[Msg] Loading SDF world file[/Users/dlaboy/Desktop/college/roboboat_stuff/repos/gz_ws/src/ardupilot_gazebo/worlds/iris_runway.sdf].
[Msg] Serving entity system service on [/entity/system/add]
[Dbg] [Physics.cc:860] Loaded [gz::physics::dartsim::Plugin] from library [/opt/homebrew/Cellar/gz-physics7/7.0.0_4/lib/gz-physics-7/engine-plugins/libgz-physics-dartsim-plugin.dylib]
[Dbg] [SystemManager.cc:74] Loaded system [gz::sim::systems::Physics] for entity [1]
[Dbg] [Sensors.cc:527] Configuring Sensors system
[Dbg] [Sensors.cc:446] SensorsPrivate::Run
[Dbg] [Sensors.cc:422] SensorsPrivate::RenderThread started
[Dbg] [SystemManager.cc:74] Loaded system [gz::sim::systems::Sensors] for entity [1]
[Dbg] [Sensors.cc:262] Waiting for init
[Msg] Create service on [/world/iris_runway/create]
[Msg] Remove service on [/world/iris_runway/remove]
[Msg] Pose service on [/world/iris_runway/set_pose]
[Msg] Pose service on [/world/iris_runway/set_pose_vector]
[Msg] Light configuration service on [/world/iris_runway/light_config]
[Msg] Physics service on [/world/iris_runway/set_physics]
[Msg] SphericalCoordinates service on [/world/iris_runway/set_spherical_coordinates]
[Msg] Enable collision service on [/world/iris_runway/enable_collision]
[Msg] Disable collision service on [/world/iris_runway/disable_collision]
[Msg] Material service on [/world/iris_runway/visual_config]
[Msg] Material service on [/world/iris_runway/wheel_slip]
[Dbg] [SystemManager.cc:74] Loaded system [gz::sim::systems::UserCommands] for entity [1]
[Dbg] [SystemManager.cc:74] Loaded system [gz::sim::systems::SceneBroadcaster] for entity [1]
[Dbg] [SystemManager.cc:74] Loaded system [gz::sim::systems::Imu] for entity [1]
[Dbg] [SystemManager.cc:74] Loaded system [gz::sim::systems::NavSat] for entity [1]
[Dbg] [SystemManager.cc:74] Loaded system [gz::sim::systems::JointStatePublisher] for entity [15]
[Dbg] [SystemManager.cc:74] Loaded system [gz::sim::systems::LiftDrag] for entity [15]
[Dbg] [SystemManager.cc:74] Loaded system [gz::sim::systems::LiftDrag] for entity [15]
[Dbg] [SystemManager.cc:74] Loaded system [gz::sim::systems::LiftDrag] for entity [15]
[Dbg] [SystemManager.cc:74] Loaded system [gz::sim::systems::LiftDrag] for entity [15]
[Dbg] [SystemManager.cc:74] Loaded system [gz::sim::systems::LiftDrag] for entity [15]
[Dbg] [SystemManager.cc:74] Loaded system [gz::sim::systems::LiftDrag] for entity [15]
[Dbg] [SystemManager.cc:74] Loaded system [gz::sim::systems::LiftDrag] for entity [15]
[Dbg] [SystemManager.cc:74] Loaded system [gz::sim::systems::LiftDrag] for entity [15]
[Msg] ApplyJointForce subscribing to Double messages on [/model/iris_with_ardupilot/joint/iris_with_standoffs::rotor_0_joint/cmd_force]
[Dbg] [SystemManager.cc:74] Loaded system [gz::sim::systems::ApplyJointForce] for entity [15]
[Msg] ApplyJointForce subscribing to Double messages on [/model/iris_with_ardupilot/joint/iris_with_standoffs::rotor_1_joint/cmd_force]
[Dbg] [SystemManager.cc:74] Loaded system [gz::sim::systems::ApplyJointForce] for entity [15]
[Msg] ApplyJointForce subscribing to Double messages on [/model/iris_with_ardupilot/joint/iris_with_standoffs::rotor_2_joint/cmd_force]
[Dbg] [SystemManager.cc:74] Loaded system [gz::sim::systems::ApplyJointForce] for entity [15]
[Msg] ApplyJointForce subscribing to Double messages on [/model/iris_with_ardupilot/joint/iris_with_standoffs::rotor_3_joint/cmd_force]
[Dbg] [SystemManager.cc:74] Loaded system [gz::sim::systems::ApplyJointForce] for entity [15]
[Dbg] [SystemManager.cc:74] Loaded system [ArduPilotPlugin] for entity [15]
[Msg] Loaded level [3]
[Msg] Serving world controls on [/world/iris_runway/control], [/world/iris_runway/control/state] and [/world/iris_runway/playback/control]
[Msg] Serving GUI information on [/world/iris_runway/gui/info]
[Msg] World [iris_runway] initialized with [1ms] physics profile.
[Msg] Serving world SDF generation service on [/world/iris_runway/generate_world_sdf]
[Msg] Serving world names on [/gazebo/worlds]
[Msg] Resource path add service on [/gazebo/resource_paths/add].
[Msg] Resource path get service on [/gazebo/resource_paths/get].
[Msg] Resource path resolve service on [/gazebo/resource_paths/resolve].
[Msg] Resource paths published on [/gazebo/resource_paths].
[Msg] Server control service on [/server_control].
[Msg] Found no publishers on /stats, adding root stats topic
[Msg] Found no publishers on /clock, adding root clock topic
[Dbg] [SimulationRunner.cc:545] Creating PostUpdate worker threads: 7
[Dbg] [SimulationRunner.cc:556] Creating postupdate worker thread (0)
[Dbg] [SimulationRunner.cc:556] Creating postupdate worker thread (1)
[Dbg] [SimulationRunner.cc:556] Creating postupdate worker thread (2)
[Dbg] [SimulationRunner.cc:556] Creating postupdate worker thread (3)
[Dbg] [SimulationRunner.cc:556] Creating postupdate worker thread (4)
[Dbg] [SimulationRunner.cc:556] Creating postupdate worker thread (5)
[Msg] Found IMU sensor with name [iris_with_standoffs::imu_link::imu_sensor]
[Dbg] [ArduPilotPlugin.cc:1156] Computed IMU topic to be: world/iris_runway/model/iris_with_ardupilot/model/iris_with_standoffs/link/imu_link/sensor/imu_sensor/imu
[... cut ...]
[Msg] Publishing pose messages on [/world/iris_runway/pose/info]
[Msg] Publishing dynamic pose messages on [/world/iris_runway/dynamic_pose/info]
[Dbg] [EntityComponentManager.cc:1655] Updated state thread iterators: 10 threads processing around 5 entities each.
[Wrn] [Model.hh:69] Skipping serialization / deserialization for models with //pose/@relative_to attribute.
[Wrn] [Component.hh:144] Trying to serialize component with data type [N3sdf3v145WorldE], which doesn't have `operator<<`. Component will not be serialized.

3. I tried this, still same output

closed 2024-02-29 by srmainwaring