srmainwaring/asv_wave_sim #160: Water patch is too small?

PerFrivik, 2024-03-03:
[image]

When I go too fast, this happens quite often, does anyone else get this too? 

Thanks!

PerFrivik, 2024-03-03:
Okay, after further investigation, it is not even necessarily when going fast, but usually, when I drive into a wave and the boat collision seems to penetrate the wave, the vehicle I am using is the default wam-m boat.

srmainwaring, 2024-03-03:
Are you using the wam-v with the custom collisions or another version?

PerFrivik, 2024-03-03:
Wow your simulation is really smooth, are you running in on a GPU? 

> Are you using the wam-v (from the asv_wave_sim repo) with the custom collisions or another version?

I am using the custom version from this repository. I'll try to figure out what is causing this issue. From the error I get, I believe it has something to do with the ODE.

srmainwaring, 2024-03-03:
> Wow, your simulation is really smooth, are you running it on a GPU?

Running natively on macOS (Metal rendering). I'll check on an Ubuntu 22.04 VM, but will only have software rendering in that case. There must be something odd going on if the water patch is not sized correctly - it relies on the collision mesh AABB to set this, so if the collision library is not working correctly that could explain (I did notice the aabb warning messages in the log you posted, but was hard to read - a text grab of the entire log would be handy).

PerFrivik, 2024-03-03:
I'll get the whole text next time, thank you again for all the support. It might also have something to do with the physics parameters that I have set for the world. 

    <physics type="ode">
      <max_step_size>0.004</max_step_size>
      <real_time_factor>1.0</real_time_factor>
      <real_time_update_rate>250</real_time_update_rate>
    </physics>

PerFrivik, 2024-03-03:
```
ODE INTERNAL ERROR 1: assertion "aabbBound >= dMinIntExact && aabbBound < dMaxIntExact" failed in collide() [collision_space.cpp:460]
Stack trace (most recent call last):
#31   Object "gz sim --verbose=1 -r -s /home/per/Workspaces/auterion_px4_ws/src/PX4_firmware_private/Tools/simulation/gz/worlds/waves.sdf", at 0x560ff30a51c4, in _start
#30   Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7f26a0229e3f, in __libc_start_main
#29   Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7f26a0229d8f, in 
#28   Object "gz sim --verbose=1 -r -s /home/per/Workspaces/auterion_px4_ws/src/PX4_firmware_private/Tools/simulation/gz/worlds/waves.sdf", at 0x560ff30a517e, in 
#27   Object "/lib/x86_64-linux-gnu/libruby-3.0.so.3.0", at 0x7f26a06a8e19, in ruby_run_node
#26   Object "/lib/x86_64-linux-gnu/libruby-3.0.so.3.0", at 0x7f26a06a5317, in 
#25   Object "/lib/x86_64-linux-gnu/libruby-3.0.so.3.0", at 0x7f26a083a30c, in rb_vm_exec
#24   Object "/lib/x86_64-linux-gnu/libruby-3.0.so.3.0", at 0x7f26a0834c96, in 
#23   Object "/lib/x86_64-linux-gnu/libruby-3.0.so.3.0", at 0x7f26a0831fc5, in 
#22   Object "/lib/x86_64-linux-gnu/libruby-3.0.so.3.0", at 0x7f26a082fc34, in 
#21   Object "/usr/lib/x86_64-linux-gnu/ruby/3.0.0/fiddle.so", at 0x7f26a04af44b, in 
#20   Object "/lib/x86_64-linux-gnu/libruby-3.0.so.3.0", at 0x7f26a07fd088, in rb_nogvl
#19   Object "/usr/lib/x86_64-linux-gnu/ruby/3.0.0/fiddle.so", at 0x7f26a04aed6b, in 
#18   Object "/lib/x86_64-linux-gnu/libffi.so.8", at 0x7f26a046d492, in 
#17   Object "/lib/x86_64-linux-gnu/libffi.so.8", at 0x7f26a0470e2d, in 
#16   Object "/usr/lib/x86_64-linux-gnu/libgz-sim7-gz.so.7.7.0", at 0x7f26a00d08dd, in runServer
#15   Object "/lib/x86_64-linux-gnu/libgz-sim7.so.7", at 0x7f269b53a4f5, in 
#14   Object "/lib/x86_64-linux-gnu/libgz-sim7.so.7", at 0x7f269b54960a, in gz::sim::v7::SimulationRunner::Run(unsigned long)
#13   Object "/lib/x86_64-linux-gnu/libgz-sim7.so.7", at 0x7f269b548cd0, in gz::sim::v7::SimulationRunner::Step(gz::sim::v7::UpdateInfo const&)
#12   Object "/lib/x86_64-linux-gnu/libgz-sim7.so.7", at 0x7f269b547412, in gz::sim::v7::SimulationRunner::UpdateSystems()
#11   Object "/usr/lib/x86_64-linux-gnu/gz-sim-7/plugins/libgz-sim-physics-system.so", at 0x7f268845dfc6, in gz::sim::v7::systems::Physics::Update(gz::sim::v7::UpdateInfo const&, gz::sim::v7::EntityComponentManager&)
#10   Object "/usr/lib/x86_64-linux-gnu/gz-sim-7/plugins/libgz-sim-physics-system.so", at 0x7f268845e641, in gz::sim::v7::systems::PhysicsPrivate::Step(std::chrono::duration<long, std::ratio<1l, 1000000000l> > const&)
#9    Object "/usr/lib/x86_64-linux-gnu/gz-physics-6/engine-plugins/libgz-physics-dartsim-plugin.so", at 0x7f2665796e3b, in gz::physics::dartsim::SimulationFeatures::WorldForwardStep(gz::physics::Identity const&, gz::physics::SpecifyData<gz::physics::RequireData<gz::physics::WorldPoses>, gz::physics::ExpectData<gz::physics::ChangedWorldPoses, gz::physics::Contacts, gz::physics::JointPositions> >&, gz::physics::CompositeData&, gz::physics::ExpectData<gz::physics::ApplyExternalForceTorques, gz::physics::ApplyGeneralizedForces, gz::physics::VelocityControlCommands, gz::physics::ServoControlCommands, std::chrono::duration<long, std::ratio<1l, 1000000000l> > > const&)
#8    Object "/lib/x86_64-linux-gnu/libdart.so.6.12", at 0x7f2665366e73, in dart::simulation::World::step(bool)
#7    Object "/lib/x86_64-linux-gnu/libdart.so.6.12", at 0x7f266534c705, in dart::constraint::ConstraintSolver::solve()
#6    Object "/lib/x86_64-linux-gnu/libdart.so.6.12", at 0x7f266534afe1, in dart::constraint::ConstraintSolver::updateConstraints()
#5    Object "/lib/x86_64-linux-gnu/libdart-collision-ode.so.6.12", at 0x7f266ae9a1cb, in dart::collision::OdeCollisionDetector::collide(dart::collision::CollisionGroup*, dart::collision::CollisionOption const&, dart::collision::CollisionResult*)
#4    Object "/lib/x86_64-linux-gnu/libode.so.8", at 0x7f266871e266, in dxHashSpace::collide(void*, void (*)(void*, dxGeom*, dxGeom*))
#3    Object "/lib/x86_64-linux-gnu/libode.so.8", at 0x7f26687267b7, in dDebug
#2    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7f26a02287f2, in abort
#1    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7f26a0242475, in raise
#0    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7f26a02969fc, in pthread_kill
Aborted (Signal sent by tkill() 848487 1001)

```

srmainwaring, 2024-03-03:
Ok - that looks like an object has exited the world bounding box due to an unreasonable acceleration. Could be time-stepping. The default is

```xml
<physics type="ode">
  <max_step_size>0.001</max_step_size>
</physics>
```

Does that make any difference? Also, can you post the entire log from startup. There may be some hints in the loading messages.

PerFrivik, 2024-03-03:
I tried: 

    <physics type="ode">
      <max_step_size>0.002</max_step_size>
      <real_time_factor>1.0</real_time_factor>
      <real_time_update_rate>500</real_time_update_rate>
    </physics>
    
   And this works really well! No crashes so far, I don't yet quite understand why though.

PerFrivik, 2024-03-03:
I'll test some more, with also higher speeds and let you know how it goes!

AntonioViscomi, 2024-05-03:
> I tried:
> 
> ```
> <physics type="ode">
>   <max_step_size>0.002</max_step_size>
>   <real_time_factor>1.0</real_time_factor>
>   <real_time_update_rate>500</real_time_update_rate>
> </physics>
> ```
> 
> And this works really well! No crashes so far, I don't yet quite understand why though.

Where did you find this parameter? I am trying to use “waves.sdf”, but I encounter the same issue.