srmainwaring/asv_wave_sim #164: problems running on Ubuntu 24.04

OlegARinchov, 2024-05-25:
When using Ubuntu 24.04 LTS with ROS2 Jazzy and Gazebo Harmonic, the asv_wave_sim plugin is compiling without errors (however, there are multiple warnings which seem non-critical though). It also launches in Gazebo and the forces seem to be applied to floating objects correctly.
However, the wavefield is rendered static and the ocean surface isn't moving in Gazebo. There seems to be some problem wuth the rendering and loading of the OGRE2 engine libs:

[GUI] [Msg] Loading plugin [gz-waves1-rendering-ogre2]
Error while loading the library [/home/superhuman/gz_ws/install/lib/libgz-waves1-rendering-ogre2.so]: libOgreNextMain.so.2.3.1: cannot open shared object file: No such file or directory
[GUI] [Err] [RenderEngineExtensionManager.cc:478] Failed to load plugin [gz-waves1-rendering-ogre2] : couldn't load library on path [/home/superhuman/gz_ws/install/lib/libgz-waves1-rendering-ogre2.so].

Are you planning to provide a fix for the plugin to work with this latest version of Ubuntu/ROS2?

And thanks for providing this great plugin!

srmainwaring, 2024-05-25:
@OlegARinchov it looks like the paths to the wave plugin libraries (GZ_SIM_SYSTEM_PLUGIN_PATH) may not be configured correctly.

I've tested the plugin on Ubuntu 24.04 running in VMware (on a macOS x86_64 host). Hardware acceleration has to be disabled because there are still issues running ogre-next accelerated in vmware - however the simulation appears to run as expected.

[image]

I do need to do something to reduce the build warnings however.

OlegARinchov, 2024-05-26:
Thanks for checking. I tried to include the path to the missing lib into GZ_SIM_SYSTEM_PLUGIN_PATH, but no change of behavior.
It seems the plugin libgz-waves1-rendering-ogre2.so somehow refers to libOgreNextMain.so.2.3.1, which cannot be found. Actually, at least on my system, ROS2 Jazzy uses OGRE 2.3.3 and this is also the default in Ubuntu 24.04. There is a  standard deb package to install OGRE 2.3.1 in parallel, but even after doing so and adding its path to GZ_SIM_SYSTEM_PLUGIN_PATH, the library still isn't found by Gazebo.

[image]

Maybe something is messed up with my install, although I just used the official standard packages for ROS2 Jazzy and Gazebo Harmonic that are provided for Ubuntu Noble. Which OGRE versions are installed in your setup?

srmainwaring, 2024-05-26:
Ah - there may be an issue with the two ogre-next versions in 24.04. I had to build `gz-harmonic` from source to get it working due to an `libOgreMain` library conflict (separate issue, but trying to get hardware acceleration working in Ogre next and the two system versions don't help). I may need to completely remove all elements of `gz-harmonic` and ogre-next to ensure a clean install.

Harmonic has only recently been released in 24.04, so there may still be some teething issues.

OlegARinchov, 2024-05-26:
Thanks! Looking forward to testing the plugin once the changes are implemented.

srmainwaring, 2024-05-27:
> Thanks! Looking forward to testing the plugin once the changes are implemented.

It's merged into `master` now. lmk if it's working for your environment.

OlegARinchov, 2024-05-28:
Thanks. I downloaded the latest code, cleaned up the old build and started over again. These is the output from colcon:

```
--- stderr: gz-waves1                                          
/home/superhuman/gz_ws/src/asv_wave_sim/gz-waves/src/systems/waves/WavesModel.cc:167:2: warning: extra ‘;’ [-Wpedantic]
  167 | };
      |  ^
/home/superhuman/gz_ws/src/asv_wave_sim/gz-waves/src/systems/dynamic/DynamicGeometry.cc:428:2: warning: extra ‘;’ [-Wpedantic]
  428 | };
      |  ^
/home/superhuman/gz_ws/src/asv_wave_sim/gz-waves/src/systems/dynamic/DynamicGeometry.cc:434:2: warning: extra ‘;’ [-Wpedantic]
  434 | };
      |  ^
/home/superhuman/gz_ws/src/asv_wave_sim/gz-waves/src/systems/hydrodynamics/Hydrodynamics.cc:543:2: warning: extra ‘;’ [-Wpedantic]
  543 | };
      |  ^
/home/superhuman/gz_ws/src/asv_wave_sim/gz-waves/src/systems/waves/WavesVisual.cc:518:2: warning: extra ‘;’ [-Wpedantic]
  518 | };
      |  ^
---
Finished <<< gz-waves1 [1min 37s]

Summary: 1 package finished [1min 37s]
  1 package had stderr output: gz-waves1
```

Still some warnings, but seems not to be anything critical. However, launching Gazebo with the plugin, I get the same behavior and error as before:

```
[GUI] [Msg] Loading plugin [gz-waves1-rendering-ogre2]
Error while loading the library [/home/superhuman/gz_ws/install/lib/libgz-waves1-rendering-ogre2.so]: libOgreNextMain.so.2.3.1: cannot open shared object file: No such file or directory
[GUI] [Err] [RenderEngineExtensionManager.cc:479] Failed to load plugin [gz-waves1-rendering-ogre2] : couldn't load library on path [/home/superhuman/gz_ws/install/lib/libgz-waves1-rendering-ogre2.so].
```

No idea, why it doesn't find the Ogre Next lib. It looks like ROS2 Jazzy and Gazebo Harmonic install different versions:

Gazebo: /usr/lib/x86_64-linux-gnu/OGRE-2.3/libOgreNextMain.so.2.3.1
ROS2: /opt/ros/jazzy/opt/gz_ogre_next_vendor/lib/libOgreNextMain.so.2.3.3

I believe, these are the standard paths for these libraries, it didn't change from Humble. Just the Ogre version in Jazzy seems to be newer, but Harmonic is unchanged.

OlegARinchov, 2024-05-31:
I tried a few things, but couldn't solve this problem so far. Just would like to ask whether the installation and location of the libOgreNextMain is the same in your environment or whether there is a different setup?

Another potential cause I am following up is the GPU drivers. Actually, I have a dedicated NVidia GPU in my laptop and run it with the proprietary driver. I confirmed that hardware acceleration is enabled in other applications such as glxgears. 

```
GL_RENDERER   = NVIDIA GeForce RTX 3060 Laptop GPU/PCIe/SSE2
GL_VERSION    = 4.6.0 NVIDIA 535.171.04
GL_VENDOR     = NVIDIA Corporation
```

Question: you mentioned you have some issues with hardware acceleration in your virtual machine. Might this be also a problem in case of running Ubuntu directly on the PC without a virtual machine?

srmainwaring, 2024-05-31:
> libOgreNextMain is the same in your environment

After installing ROS 2 Jazzy and Gazebo Harmonic there are two locations for Ogre2:

On an intel machine:

- `/usr/lib/x86_64-linux-gnu/Ogre2.3`
- `/usr/lib/x86_64-linux-gnu/OgreNext`

On an arm64 machine:

- `/usr/lib/aarch64-linux-gnu/Ogre2.3`
- `/usr/lib/aarch64-linux-gnu/OgreNext`

If you suspect OpenGL drivers are a problem you can try running:

```bash
LIBGL_ALWAYS_SOFTWARE=1 gz sim -v4 -r waves.sdf
```

I don't think that is the problem in your case, it looks like a library resolution issue.

OlegARinchov, 2024-05-31:
Thanks for your help. I finally got it working.
The problem was that I installed Gazebo somehow twice:

1. via `sudo apt install ros-jazzy-ros-gz` (as described [here](https://gazebosim.org/docs/garden/ros_installation))
2. via `sudo apt install gz-harmonic` (as described [here](https://gazebosim.org/docs/harmonic/install_ubuntu))

Ogre Next is installed at the following locations, depending on the installation method above:

1.  /opt/ros/jazzy/opt/gz_ogre_next_vendor/lib/libOgreNextMain.so.2.3.3
2.  /usr/lib/x86_64-linux-gnu/OGRE-2.3/libOgreNextMain.so.2.3.1

in my case it always called the version 2.3.3. in the ROS2 directory tree and I got an error. I completely uninstalled ROS2 and Gazebo and reinstalled this time, I strictly sticked with method 2 above. Then, only version 2.3.1 of Ogre Next was present on my system. After rebuilding the asv_waves1 package and setting up everything, it works now correctly and renders the waves nicely.

In conclusion: do not install Gazebo via the standard ROS method (1), but use method (2). It might help to inform about this issue in the installation instructions of this package or to change the build scripts to enable the flexibility to work with either Gazebo installation in the future.

sickopickle, 2025-07-06:
I had this exact same issue caused by the exact same reason, thanks so much!

sickopickle, 2025-07-08:
> Thanks for your help. I finally got it working. The problem was that I installed Gazebo somehow twice:
> 
>     1. via `sudo apt install ros-jazzy-ros-gz` (as described [here](https://gazebosim.org/docs/garden/ros_installation))
> 
>     2. via `sudo apt install gz-harmonic` (as described [here](https://gazebosim.org/docs/harmonic/install_ubuntu))
> 
> 
> Ogre Next is installed at the following locations, depending on the installation method above:
> 
>     1. /opt/ros/jazzy/opt/gz_ogre_next_vendor/lib/libOgreNextMain.so.2.3.3
> 
>     2. /usr/lib/x86_64-linux-gnu/OGRE-2.3/libOgreNextMain.so.2.3.1
> 
> 
> in my case it always called the version 2.3.3. in the ROS2 directory tree and I got an error. I completely uninstalled ROS2 and Gazebo and reinstalled this time, I strictly sticked with method 2 above. Then, only version 2.3.1 of Ogre Next was present on my system. After rebuilding the asv_waves1 package and setting up everything, it works now correctly and renders the waves nicely.
> 
> However, there is a lot of manual work to connect Gazebo installed by method 2 to ROS. So, I found you can also install both versions (installed via 1 and 2), but then eliminate all references to /opt/ros/jazzy/opt/gz_* in the environment variables PATH and LD_LIBRARY_PATH. In addition `export GZ_CONFIG_PATH=/usr/share/gz/` .
> 
> It's all a bit cumbersome, but it works for now. I am hoping, you can provide a version of the package cooperating natively with ROS2 Jazzy and Gazebo Harmonic (preferably installed via method 1) some day. It's a really nice package I prefer greatly over alternatives.

Is this still the method you are using, or did you find another way?

sickopickle, 2025-07-11:
@SeapathMaster Thanks! I was having issues running the [rs750](https://github.com/srmainwaring/rs750?tab=readme-ov-file), but I did a fresh install and it worked. Your CMakeLists.txt modified some of the ogre stuff right?

SeapathMaster, 2025-07-12:
I mostly modified the paths and the versions of the libraries as they also have changed moving to Gazebo Harmonic.

sickopickle, 2025-07-17:
@SeapathMaster What RTF are you getting on the simulation? I'm not sure why but after doing something my RTF decreased by a lot; I'm only getting around 0.6 RTF on the default WAM-V simulation. Previously I had installed both Gazebo Harmonic standalone and ros-gz, and had no issues. Not sure why I'm unable to reproduce this. I've tried reinstalling Ubuntu and repeating my previous steps but I'm getting a bunch of errors now. Do you know if there was some update to Gazebo that changed things?

sickopickle, 2025-07-22:
@SeapathMaster Can you please tell me the steps you took to install everything? I tried reinstalling Ubuntu and installing only ros-gz but get errors and low RTF.