srmainwaring/asv_wave_sim #168: Incompatible Gazebo libraries after update

OlegARinchov, 2024-07-02:
On Gazebo Harmonic, after updating to the latest version, I get errors of the following type:

[ruby $(which gz) sim-1] gz sim server: symbol lookup error: /usr/lib/x86_64-linux-gnu/gz-physics-7/engine-plugins/libgz-physics-dartsim-plugin.so: undefined symbol: _ZNK3sdf3v148Geometry9ConeShapeEv

Recompiling works without issues, but during runtime, Gazebo crashes with the above error.
I confirmed that the minor version of some libgz was increased after the update.

OlegARinchov, 2024-07-04:
I removed the non-standard Gazebo completely and just sticked with the ROS2 standard Gazebo installation:

`$ sudo apt install ros-jazzy-ros-gz`

srmainwaring, 2024-07-04:
Thanks @OlegARinchov - I'll take a look.

If the standalone and ROS 2 integrated Gazebo installations are inconsistent on Ubuntu 24.04 that is an issue that should br raised with OSRF. I imagine that is going to cause problems for others as well.

OlegARinchov, 2024-07-04:
I believe, it was changed intentionally given the statement in the ROS2 Jazzy release notes:
https://docs.ros.org/en/jazzy/Releases/Release-Jazzy-Jalisco.html#changes-to-how-ros-2-and-gazebo-integrate
While it might be simpler and more elegant, it also limits backwards compatibility. I suspect that this is the source of all the trouble.