srmainwaring/asv_wave_sim #169: Which spectrum is used for FFT waves?

ConnorDTaylor, 2024-07-03:
Is the P-M spectrum or ECKV spectrum used in the 'fft' wave algorithm? 

I am trying to determine which of the wave spectrums are being used, in the source code I am seeing the P-M spectrum and the ECKV spectrum. It is unclear to me which spectrum is actually being used in the simulator though. 

I thought it was the P-M spectrum with the cosine-2s spreading function, but I don't know for sure.

Connor

srmainwaring, 2024-07-03:
ECKV with ECKV spreading for standing waves or Cos2s for travelling (the default).

https://github.com/srmainwaring/asv_wave_sim/blob/40eed5158fcab37421826ffa3ba33aab94d55e28/gz-waves/src/LinearRandomFFTWaveSimulation.cc#L283-L297

ConnorDTaylor, 2024-07-03:
For my understanding: Do both of these spreading functions get used in the simulator simultaneously? Or is there a setting which defines which spreading function is used?

srmainwaring, 2024-07-03:
Looks like it's hardcoded to use the travelling wave version:

https://github.com/srmainwaring/asv_wave_sim/blob/40eed5158fcab37421826ffa3ba33aab94d55e28/gz-waves/src/LinearRandomFFTWaveSimulationImpl.hh#L213

So, `Cos2s` always.

closed 2024-07-22 by srmainwaring