srmainwaring/asv_wave_sim #170: Installation of asv_wave_sim in ROS2 Humble

adarsha-s-bharadwaj, 2024-07-30:
Hi, I'm trying to install asv_wave_sim in ubuntu 22.04 ROS2 Humble. Even though I did installation according to the readme file still I'm getting could not find gz-waves1- hydrodynamics-system shared library and even sometimes gz sim won't work only. Am I make any mistakes while installing? Please guide me through the installation part so that I have full working of waves along with the boat and gps as displayed in GitHub image.

srmainwaring, 2024-07-31:
Possibly related to https://github.com/srmainwaring/asv_wave_sim/issues/168

adarsha-s-bharadwaj, 2024-08-01:
ok I will look through it. I'm new to the boat environment and want to control the wam-v boat using keyboard, can you please tell me how to do it?

srmainwaring, 2024-08-01:
> I'm new to the boat environment and want to control the wam-v boat using keyboard, can you please tell me how to do it and how did you use ardupilot for control? any tutorial will help me.

This user guide for the BlueBoat may help setting up ArduPilot for control. The WAM-V is similar, but uses vectored thrust motors: https://github.com/ArduPilot/SITL_Models/blob/master/Gazebo/docs/BlueBoat.md

adarsha-s-bharadwaj, 2024-08-06:
Ok i will look through it. I am trying to use ocean waves to check about buoyancy in waves_wind.sdf file, but when it runs, gazebo sim7 (gazebo garden) is directly getting aborted. 

[image]

If this is not working, can you please tell me which type of wave should I use in gazebo sim to know about buoyancy and all? or how to solve this issue?

srmainwaring, 2025-01-02:
Adding a note here for future reference.

The main issue seems to be that the wave sim components do not support being added using the component inspector as per: https://github.com/gazebosim/gz-sim/pull/1549.

This is not the recommended way to load the wave system and rendering plugins, and is not a feature that has been tested. It may be supported in future, but the currently supported approach is to define and load the systems as SDF plugins.

adarsha-s-bharadwaj, 2025-05-26:
Okay Thank you so much!!