srmainwaring/asv_wave_sim #171: Knowing the wind speed that is causing the currently observed waves

ConnorDTaylor, 2024-08-16:
I am working on a forecasting model and I have been using this wave simulator for my simulation purposes.

My question is, I want to forecast what waves will be observed in the future. To do this I need the causal aspects of the waves, which would be predominantly the waves. Is there a way that I can access what the wind speed that causes the waves being observed is?

I know the wind speed (U10) is supplied by the user, so I am really wondering if this wind speed changes while the simulator is running. Perhaps there is some statistical range of wind speeds around the defined wind speed. This would make sense since the waves are seen to increase and decrease in amplitude, if the wind speed never changed I don't know if this would happen.

Is there a way to access the instantaneous wind speed during the simulation run?

ConnorDTaylor, 2024-08-16:
Looking through the source code, I can't seem to find anywhere the wind changes. Alternatively, is there a variable I can look at that will serve a similar function as a causal component of wave height changes?

srmainwaring, 2024-08-17:
Hi @ConnorDTaylor, the wind speed is a parameter of the wave spectrum function. The resulting spectrum is sampled for the amplitudes which are then feed through the 2D inverse FFT to generate the summed wave displacements. There is also an adjustment for wave spreading that results in travelling rather than standing waves. This particular model assumes a steady state sea distribution, so there isn't any physics that describes how a sea state builds, rather the model assumes the input wind parameter has been constant for long enough for the sea state (in terms of average wave height etc) to reach equilibrium.

Some notes and references in this presentation: 
https://docs.google.com/presentation/d/1JXwWMPPVT7y03Vr6cWtrwAwyvdysmg3NW9ZmI276dMY/edit?usp=sharing

ConnorDTaylor, 2024-08-18:
Hi Rhys,

I feared that would be the case based on my understanding of the ECKV spectrum. Are there any parameters, which you can think of, which could be tracked to predict how the sea state will build? 
For example, is there a way I could track the wave spreading parameter which would be similar to a changing wind direction? I imagine the seas would build to the highest amplitude when this spreading adjustment is nearly zero. is the spreading adjustment random or does it distribute cyclically?

srmainwaring, 2024-08-18:
I'm afraid the modelling of how a sea state builds is an entirely different (and very much more complicated) problem. From my understanding the steady state spectrum and spreading functions contain no dynamic information about the process by which a sea state forms from the energy transfer of wind to the sea surface.

closed 2025-01-02 by srmainwaring