srmainwaring/asv_wave_sim #174: Could not find shared library (WavesVisual)

sickopickle, 2024-12-22:
I've looked through all the other posts on this. Originally I had built it in ~/gz_ws/src/ and got errors for Hydrodynamics and other plugins as well; I think you should put something in the README to avoid this in the future. I fixed that and now those errors are gone but the WavesVisual still doesn't work. I noticed that the library search paths don't seem to be at all where the plugin actually is. Help would be greatly appreciated, thanks!

```
[GUI] [Err] [SystemLoader.cc:119] Failed to load system plugin: (Reason: Could not find shared library)
- Requested plugin name: [gz::sim::systems::WavesVisual]
- Requested library name: [gz-waves1-waves-visual-system]
- Library search paths:
  - /home/sickopickle/.gz/sim/plugins/
  - /usr/lib/x86_64-linux-gnu/gz-sim-7/plugins/
  - /home/sickopickle/.ignition/gazebo/plugins/
[GUI] [Err] [SystemLoader.cc:119] Failed to load system plugin: (Reason: Could not find shared library)
- Requested plugin name: [gz::sim::systems::WavesVisual]
- Requested library name: [gz-waves1-waves-visual-system]
- Library search paths:
  - /home/sickopickle/.gz/sim/plugins/
  - /usr/lib/x86_64-linux-gnu/gz-sim-7/plugins/
  - /home/sickopickle/.ignition/gazebo/plugins/
[GUI] [Err] [SystemLoader.cc:119] Failed to load system plugin: (Reason: Could not find shared library)
- Requested plugin name: [gz::sim::systems::WavesVisual]
- Requested library name: [gz-waves1-waves-visual-system]
- Library search paths:
  - /home/sickopickle/.gz/sim/plugins/
  - /usr/lib/x86_64-linux-gnu/gz-sim-7/plugins/
  - /home/sickopickle/.ignition/gazebo/plugins/
[GUI] [Err] [SystemLoader.cc:119] Failed to load system plugin: (Reason: Could not find shared library)
- Requested plugin name: [gz::sim::systems::WavesVisual]
- Requested library name: [gz-waves1-waves-visual-system]
- Library search paths:
  - /home/sickopickle/.gz/sim/plugins/
  - /usr/lib/x86_64-linux-gnu/gz-sim-7/plugins/
  - /home/sickopickle/.ignition/gazebo/plugins/

```

closed 2025-01-02 by srmainwaring

sickopickle, 2025-01-03:
I had previously copied and pasted the text under set environment variables and still resulted in this error. I tried deleting my gz_ws folder and trying the installation again with the same results.

srmainwaring, 2025-01-03:
Hi @sickopickle, I closed this issue because I cannot replicate on CI, or Ubuntu 22.04. It is most likely an issue with your Gazebo installation or other path.

Without further information it is difficult to identify what might be the problem with your system / configuration - listings of the `~/gz_ws/install/lib` folder may help. On macOS it is: 

```bash
% ls ./install/lib 
FreeSurfaceHydrodynamics
cmake
libasv_sim2-anemometer-system.2.0.0.dylib
libasv_sim2-anemometer-system.2.dylib
libasv_sim2-anemometer-system.dylib
libasv_sim2-foil-lift-drag-system.2.0.0.dylib
libasv_sim2-foil-lift-drag-system.2.dylib
libasv_sim2-foil-lift-drag-system.dylib
libasv_sim2-mooring-system.2.0.0.dylib
libasv_sim2-mooring-system.2.dylib
libasv_sim2-mooring-system.dylib
libasv_sim2-sail-lift-drag-system.2.0.0.dylib
libasv_sim2-sail-lift-drag-system.2.dylib
libasv_sim2-sail-lift-drag-system.dylib
libasv_sim2-sail-position-controller-system.2.0.0.dylib
libasv_sim2-sail-position-controller-system.2.dylib
libasv_sim2-sail-position-controller-system.dylib
libasv_sim2-wind-system.2.0.0.dylib
libasv_sim2-wind-system.2.dylib
libasv_sim2-wind-system.dylib
libasv_sim2.2.0.0.dylib
libasv_sim2.2.dylib
libasv_sim2.dylib
libgz-waves1-dynamic-geometry-system.1.0.0.dylib
libgz-waves1-dynamic-geometry-system.1.dylib
libgz-waves1-dynamic-geometry-system.dylib
libgz-waves1-hydrodynamics-system.1.0.0.dylib
libgz-waves1-hydrodynamics-system.1.dylib
libgz-waves1-hydrodynamics-system.dylib
libgz-waves1-rendering-ogre2.1.0.0.dylib
libgz-waves1-rendering-ogre2.1.dylib
libgz-waves1-rendering-ogre2.dylib
libgz-waves1-rendering.1.0.0.dylib
libgz-waves1-rendering.1.dylib
libgz-waves1-rendering.dylib
libgz-waves1-waves-model-system.1.0.0.dylib
libgz-waves1-waves-model-system.1.dylib
libgz-waves1-waves-model-system.dylib
libgz-waves1-waves-visual-system.1.0.0.dylib
libgz-waves1-waves-visual-system.1.dylib
libgz-waves1-waves-visual-system.dylib
libgz-waves1.1.0.0.dylib
libgz-waves1.1.dylib
libgz-waves1.dylib
pkgconfig
```

sickopickle, 2025-03-06:
I figured out the issue. I didn't set the environment variables for the second terminal window in which I ran `gz sim -v4 -g`. It works now. Thanks!