srmainwaring/asv_wave_sim #178: Integration of PX4 drone

sickopickle, 2025-03-07:
I'm pretty new to Gazebo and this stuff in general. I'm trying to add a PX4 drone into the environment and land it on the WAM-V. How can I do this? I've installed PX4 using the instructions on their website.

Thanks!

srmainwaring, 2025-03-07:
@sickopickle I'm afraid that's not a question that's directly related to this repo. PX4 support is available on their discuss and discord servers. I've provided Gazebo examples for landing quad planes on ships using ArduPilot here: https://discuss.ardupilot.org/t/ship-landing-ignition-simulation/80663.

closed 2025-03-07 by srmainwaring

sickopickle, 2025-03-10:
I got it to work, if anyone is interested.

Here are the steps:

1. Install PX4 https://docs.px4.io/main/en/dev_setup/dev_env_linux_ubuntu.html 
2. Install asv_wave_sim; build wave control plugin
3. Place the gz_ws folder into PX4-Autopilot folder.
4. Add [this modified waves.sdf file](https://pastebin.com/ududh4dR) to the /home/pickle/PX4-Autopilot/Tools/simulation/gz/worlds folder
5. cd PX4-Autopilot/gz_ws
6. source ./install/setup.bash
7. Paste this modified "set environment variables"

# for future use - to support multiple Gazebo versions
export GZ_VERSION=harmonic

# not usually required as should default to localhost address
export GZ_IP=127.0.0.1

# ensure the model and world files are found
export GZ_SIM_RESOURCE_PATH=\
$GZ_SIM_RESOURCE_PATH:\
$HOME/PX4-Autopilot/gz_ws/src/asv_wave_sim/gz-waves-models/models:\
$HOME/PX4-Autopilot/gz_ws/src/asv_wave_sim/gz-waves-models/world_models:\
$HOME/PX4-Autopilot/gz_ws/src/asv_wave_sim/gz-waves-models/worlds

# ensure the system plugins are found
export GZ_SIM_SYSTEM_PLUGIN_PATH=\
$GZ_SIM_SYSTEM_PLUGIN_PATH:\
$HOME/PX4-Autopilot/gz_ws/install/lib

# ensure the gui plugin is found
export GZ_GUI_PLUGIN_PATH=\
$GZ_GUI_PLUGIN_PATH:\
$HOME/PX4-Autopilot/gz_ws/src/asv_wave_sim/gz-waves/src/gui/plugins/waves_control/build

8. Install QGroundControl https://docs.px4.io/main/en/dev_setup/qgc_daily_build.html 
9. Run ./QGroundControl.AppImage and PX4_GZ_WORLD=waves make px4_sitl gz_x500 in separate terminals. 

(this took me a while to figure out)